AllskyTeam/allsky #174: Meteors vs Airplanes

phil-zenger, 2020-04-21:
Last night (4/20/2020) was the first night i put my allsky camera out. This question will sound stupid, but how do I tell the difference between meteors and airplanes. Can somebody watch my mp4 and tell me what you see?

Here is my Youtube link to the video:

https://youtu.be/9_S5fOakML0

kpjamro, 2020-04-21:
Except for the really big ones, meteors rarely last more than 1 second.  In a series of 10-second exposures you will only see a meteor in 2 successive frames if it hits right at the frame boundary.  Since each of the exposures is an average of 10 seconds, meteors need to be pretty bright to show up (like 10x the brightness of the background sky).  The majority of them are rather faint and require different software/hardware to detect.

So: anything you see in 3 or more successive frames is _certainly_ an airplane and, if it is in 2 frames, it is _possibly_  an airplane.

Seeing something that lasts for only one frame in a video playback is rather difficult.  But I think I saw two in your video: one at 26 seconds and one at ~1:22 - both on the right side of the picture (3 o'clock position). Don't blink or you will miss them.

Meteors that are coming straight at you do not leave much of a trail - just a blink of light.

There should be more meteors tonight!

Ken

phil-zenger, 2020-04-21:
Ken,

Thanks for you quick response. That gives me a better idea of what to look for.

closed 2020-04-21 by phil-zenger

kpjamro, 2020-04-21:
you may also see Starlink satellites. They move pretty fast - but, of course, slower than meteors.

The attached is from my GMS meteor cam. There is one (faint) meteor and several of the satellites.

[image]