AllskyTeam/allsky #175: Stylesheet not interpreted

rogergau, 2020-04-26:
The stylesheet doesn't seem to be taking effect on my Pi4 after upgrading. Not sure if it's an issue with AllSky or with updates on the Pi. I updated AllSky and a full apt-get update also.

`Resource interpreted as Stylesheet but transferred with MIME type application/octet-stream:`

[image]

I'm guessing it's something specific to my Pi as I see no other references to this for All Sky ...

Any tips?
Roger.

thomasjacquin, 2020-04-26:
Others have had that issue. I am not too sure what is exactly happening but I would be interested to find out so /i can fix it for good.
https://github.com/thomasjacquin/allsky/issues/110

rogergau, 2020-04-26:
Thanks. FYI - I had done the git/install.sh mentioned in that thread. Also cleared cache. Those hadn't helped. I then tried a reboot but unfortunately the Pi hasn't come back online following that so I'll have to put this on ice for a week or two until I'm next on-site to power cycle the Pi. :(

gereta, 2021-03-19:
Hmm... I experienced the same issue. I read somewhere that line 
#mimetype.assign             = (".css" => "text/css", )
in /etc/lighttpd/lighttpd.conf should be commented and after install it was.
After removing trailing # (uncomment) of this line and restart: sudo service lighttpd force-reload
it works as it should.

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys