AllskyTeam/allsky #178: Daytime Interval also applies to night time recording

locastan, 2020-05-08:
Hi Thomas,

if I set a longer daytime recording interval for weather camera purposes through the web GUI, it seems to also use that one for night time recording. (I am however not sure I am running the latest version of your project, because I added dew control to my version. So it's possible you fixed this already, let me know if thats the case.)

Grettings, Mario.

thomasjacquin, 2020-05-14:
Daytime and night time interval should be separate but you could see side effects if the daytime delay is very long or if the RPi time is wrong.

locastan, 2020-05-14:
Pi clock is correct. How long is very long? I set it to a minute, so there would be less thermal stress to the camera during daytime.

thomasjacquin, 2020-05-14:
I was talking about things like 1 hour.. Then in that case, the loops is stuck for an hour before checking for night time again.

locastan, 2020-05-14:
OK. I will have to check why it went into 1 minute intervals between exposures for night recording because of that setting. The camera sensor is reaching 80+°C during sunny spring days now, so I have to do something.

EricClaeys, 2021-09-14:
This issue is from an older release of the AllSky software and a suggested solution provided.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks

closed 2021-09-14 by EricClaeys