AllskyTeam/allsky #179: Keogram and startrail white

bendrewery, 2020-05-13:
Does anyone know why both my keograms and startrails generate only a white image. The keogram does have some grain effect to the image, possibly showing some start, but totally inverted blank/white.

I have tried different values for the rejection but 0, 1 or any value between generates the same result.
I am using an ASI178MM.

Any help apreciated.

Ben

phil-zenger, 2020-05-13:
In your settings.json file, what is your setting for TYPE? Also, in your config.sh what is your setting for BRIGHTNESS_THRESHOLD?

bendrewery, 2020-05-13:
My Type = 2 16bit RAW. Brightness_threshold I have tried many, the default
or even the extremes doesn't seem to make a difference. I also notice that
no overlay seems to work with the 16bit raw, and pretty flakey with 8bit
raw.

phil-zenger, 2020-05-13:
Your choices for Type are as follows: Image format. 0=RAW 8 bits, 1=RGB 24 bits, 2=RAW 16 bits. When I have chosen TYPE = 2, it produces only white images. I would change your TYPE =1.  

As for your setting for startrails, I would run the process manually a few times so you can understand what it is doing. Here is how to run it manually:

from the /home/pi/allsky directory, run (change your date folder to one with actual images)
./startrails ./images/20200428/ jpg 0.18 ./images/20200428/startrails/startrails_2.jpg

As the process runs, it will show you the brightness value for each image. Based on the conditions, like clouds or moon etc, you will most likely have to up the brightness threshold. 

Run the processes a 2nd time. Like this:

./startrails ./images/20200428/ jpg 0.25 ./images/20200428/startrails/startrails_3.jpg

You might have to run it a third time, like this:

./startrails ./images/20200428/ jpg 0.35 ./images/20200428/startrails/startrails_3.jpg

bendrewery, 2020-05-15:
Using Type 1, RGB 24bit has got it all working as normal. So the startrails and keograms generate properly.

Thank you for the manual command, very useful for figuring out the optimal value.

closed 2020-05-15 by bendrewery