AllskyTeam/allsky #182: Feature Request: Better Dark Capture

robertpascale, 2020-05-16:
Dark frames are a function of exposure length at certain gains at a given temperature.
Current Dark frame implementation requires a "snapshot in time" where the temperature and exposures are congruent with "current conditions". However, this is unlikely to work well overall as too many of the variable parameters are moving (temperature, exposure length). Attempting to control these variables (or compensate for them) would be beneficial in creating lower noise images.

The suggested feature request is an admin function to create a dark library for the camera.
For non-cooled cameras, this may be simply a set of dark frames taken at x-second intervals for each duration could be 1,2,5 or 10 - eg. DARK01s, DARK02s,....DARK60s. Most exposures will not exceed 1 minute even on the current lower specification cameras.
When the images are being taken, subtraction of the dark frame with the closest duration to the current exposure would be used for the dark subtraction (eg. 14.7s exposure would pick a 15sec DARK for subtraction)
For cases where the camera is cooled, the dark frame library creation routine should attempt to cool to a certain level before taking the dark library frames.

curitsreno, 2020-07-11:
robertpascale, 
I am working on a dynamic solution to the dark frame static image today. Will be testing it tonight and ill let you know.
---- 
Some Background and my idea of a fix:

After looking over the code, I think I have a good idea of how to patch in dynamic dark frames,

But basically from what I can tell:
The capture program just captures, writes to log then calls script depending on day / night / etc. 
The  saveImageNight.sh script does the dark.jpg subtraction after the capture finishes with its current capture. 
The log file just records capture events, filename, exposure, gain, etc per capture.

First problem - need to know exposure before conversion - it's in the log, but not logged in a way that says hey that's an exposure value. To fix that I added 2 lines of code and changed one line for the logging routine in the capture code to perpend exposures with 'EXP'. Also had to adjust the call to logging for the exposure value, just changed one line down further in the capture code. (capture.cpp code: added 2 lines of code and changed 2 lines)

Second problem - the saveImageNight.sh script only calls 1 dark image. To fix that I added two lines in the dark frame conversion portion of the script. Basically, it reads the log file and finds the last reference of EXP and extracts the exposure number. Then rounds to nearest 1 millionth because 1 second of exposure is saved as 1000000 in the log. Then adds the exposure number to the dark.png file used for dark conversion. So a 3.500 to 4.499 second exposure would make the script look for dark4.png to convert with.  (saveImageNight.sh code: added 2 lines)

So now that the Allsky program is looking for different exposure frames we just need to make dark frame files for every second of prolonged exposure. If your max exposure is say 20 seconds then you need to make 21 dark frame files. 0 for under 1 second all the way up to 20. 
Set your max exposure to 0.499. Then run the dark frame routine. Save file as dark0.png  
Set your max exposure to 1.   Then run dark frame routine. Save file as dark1.png
Set your max exposure to 2.   Then run dark frame routine. Save file as dark2.png
And so on till your at max exposure setting.
NOTE: you will have to manually copy your dark.png to dark(n).png between each dark frame exposure sessions.

Hope you were able to follow that, Until I test tonight I won't know if it works. But I will let you know...

Note this is for my ASI120MC camera

svego, 2021-01-24:
Good idea! But why not create a script that will generate all the darks?

EricClaeys, 2021-09-14:
The current dark frame mechanism is temperature based, not exposure based.  I'd guess the assumption is that most night-time shots have similar exposure times (except at the start and end of night, but those exposure times might be too short to warrant using dark frames), but different temperatures.
Can this issue be closed?  If you still plan to propose implementing time-based dark frames you'll want to use the version 0.8 code which improved dark capture.

EricClaeys, 2021-10-16:
@robertpascale, @svego,
The current dark frame mechanism is temperature based, not exposure based. I'd guess the assumption is that most night-time shots have similar exposure times (except at the start and end of night, but those exposure times might be too short to warrant using dark frames), but different temperatures.

Can this issue be closed? If you still plan to propose implementing time-based dark frames you'll want to use the version 0.8 code which improved dark capture.  Also, a future version of allsky will combine saveDayImage.sh and saveNightImage.sh into saveImage.sh, and the capture programs will call saveImage.sh, passing in the temperature, so the temperature.txt file will no longer be needed.  The capture program could easily pass the exposure time as well, which would simplify things.