AllskyTeam/allsky #187: Allsky service issue

WirthmU, 2020-05-26:
I had to re-setup for other reasons after Allsky was working perfectly well for quite come time. I did update and upgrade and rpi-update before re-installing Allsky:
raspberry Pi 4 Model B Rev 1.1
Raspbian GNU/Linux 10 (buster)

Now I get a startup issue. Allsky and its service has been installed as usual and after rebooting I get:

sudo service allsky status:
`
allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Mon 2020-05-25 12:00:27 CEST; 20h ago
  Process: 555 ExecStart=/home/pi/allsky/allsky.sh (code=killed, signal=TERM)
 Main PID: 555 (code=killed, signal=TERM)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/allsky.service

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
`
If I do start Allsky by hand it works perfectly fine.

Any help to this?

thomasjacquin, 2020-06-08:
Sorry for the late reply,

Have you figured this out yet? Is there anything useful in `/var/log/allsky.log`?

WirthmU, 2020-06-08:
Thank you for your reply.

To answer your question, there was nothing really helpful in allsky.log - the camera just stopped taking pictures without any notion of a reason in the log file. Re-starting the service made it go on for a few minutes or hours again. I might not have remembered well my previous Parametern in the settings.json file and thus overloaded my RPI4 with the ASI385MC camera data.

Now binning is set to two and quality to 80 and the allsky camera has been online and working for four days in a row.

Are there any RPI4 configuration or settings.json parameters that can be optimized so I could use bin 1?

WirthmU, 2020-06-12:
The camera stopped again taking pictures after 5 days of continuous running.
`Allsky service status claims that the camera is still running:
llsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-06-06 12:12:52 CEST; 5 days ago
 Main PID: 588 (allsky.sh)
    Tasks: 6 (limit: 4915)
   CGroup: /system.slice/allsky.service
           ├─588 /bin/bash /home/pi/allsky/allsky.sh
           └─639 ./capture -angle -6 -autoexposure 1 -autogain 0 -bin 2 -brightness 50 -coolerEnabled 0 -darkframe 0 -daytimeDelay 50000 -delay 10 -exposure 10000 -file

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
`
The /var/log/allsky.log stopped logging already three days ago although the camera continued to work until last night at around  10pm.

Restarting the service made the camera immediately working again.

thomasjacquin, 2020-06-18:
This seems like a pretty random issue. On cause could be a failing SD card. Sometimes blocks get corrupted and cause instability in the system. I've had this happen in the past.

WirthmU, 2020-07-26:
I am still intrigued by the allsky system halt without any errors or failure notification in any log files and the allsky service pretending to be up and running. Simply restart the service restores the camera function immediately.

The logfile in the allsky directory has become rather big by now. Is there any setting which would suppress the generation of this log file?

Dentonknifeworks, 2020-07-26:
I just edit it and clear the contents of it then save it.

EricClaeys, 2021-09-16:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  If you are still having this problem with version 0.8, try increasing the log level to its highest value, which may give you an idea if it's software related.
The "signal=TERM" is what the service daemon sends to the allsky.sh program to stop it.  Not sure if that's related here.

closed 2021-09-16 by EricClaeys