AllskyTeam/allsky #188: Feature Request: Image crop/image circle

pterodyne, 2020-05-26:
It would be nice to crop to a square for those of us who have circular fisheye lenses to remove all the unused pixels.  Secondly the image would be cleaner if there was an option to mask out the area outside of that image circle.

Thanks for listening.

Alex-developer, 2020-08-12:
You can do this yourself, assuming you have some basic knowledge of scripting. I crop my images at the beginning of the saveimagefornight script so that the video, star trails etc all use it.

convert "$IMAGE_TO_USE" -gravity Center -crop 2048x2048+0+50 +repage "$IMAGE_TO_USE";

I was thinking of adding this as a configurable option along with a few others. If I get some free time at the weekend I will have a go at this and do a Pull Request for Thomas

Alex-developer, 2020-08-16:
I have just submitted a Pull request https://github.com/thomasjacquin/allsky/pull/223 to add cropping via the config.

Alex-developer, 2020-08-16:
@thomasjacquin has merged the Pull request so this is now available in the main code @pterodyne

pterodyne, 2020-08-17:
Thanks playing with it.

pterodyne, 2020-08-17:
works for me.  The only issue I see is it crops off the image details.  Im assuming if I set the offset it will work.  Trying that now.

pterodyne, 2020-08-17:
Yes, so if you enable this feature, please remember to set the text offset to compensate for the crop.

Alex-developer, 2020-08-17:
> Yes, so if you enable this feature, please remember to set the text offset to compensate for the crop.

Sorry yes I should have mentioned that !