AllskyTeam/allsky #190: PI 2 going to sleep?

perulero, 2020-05-26:
Hi

My Pi 2 is running stretch linux and has the Allsky software installed. Installed recently from a stored image made in 2018.
My issue is everything works fine with my ASI1034 camera but at the end of the night it seems to shut down or go to sleep?
I try and loging using VNC or ALLSKY.LOCAL and cannot ?
It is off the network until I cycle the power to it?
Is there a setting I can change to stop it going to slepp?

thomasjacquin, 2020-06-08:
I am not sure it is going to sleep. There might be something else going on. Try connecting to it using SSH or http://your-ip-address
If you connect with SSH, you can monitor the running processes by using the command `top` or `ps -aux`. You should see `./capture` and `timelapse.sh` at the end of the night.

minnican, 2020-07-02:
I had to upgrade my power supply from 5V 2.0A to 5V **3.0A** as my Pi and the allsky software was hanging at random times during the day and night. It's seems that the camera attached to the Pi were drawing too much power together. Since upgrading I've had no problems.

EricClaeys, 2021-09-16:
Issue resolved by user.

closed 2021-09-16 by EricClaeys