AllskyTeam/allsky #193: black image during daylite exposure

The-J-BB, 2020-05-31:
Since yesterday, I got several times only black images with increasing exposure times, during daytime. It seems to be a comunucation problem of the USB lane, because I changed the USB cable from a commont type 2.0 to 3.0. Maybe it also could be a temperature problem o the sensor but I don't think, because a day before I had sensor tempereatures almost 20° more and anything worked well. Now I will try to cool down the sensor as well as change the USB cable back to 2.0 version. I will report the results in this thread.
If anybody had the same problems, please post your experiences and maybe solutions. I use the latest Raspberian version and a Rasperry Pi 4.

The-J-BB, 2020-05-31:
I should additionally note that the CPU usage while showing these black images is 0%, so nothing is processing or one or more processes are not running. The used camera is ZWO ASI 178MC.

dbainbridge, 2020-06-05:
As a data point I am using same camera on a Pi 3b+ and according to my current day time image it says the sensor is 60C!  (and still appears to be working fine).

EricClaeys, 2021-09-16:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.

closed 2021-09-16 by EricClaeys