AllskyTeam/allsky #194: Gain settings not being applied after install

cjdavis618, 2020-06-07:
ASI 120MC-S - Connected to RPi4 with POE Shield for power. All setup today. 
Fresh load of Raspbian, updated OS and followed instructions on this Git page to install using scripts. 
After main install, Gui was installed as well so I could watch the feed remotely. 

At first, I had banding issues and some blocky frames with image corruption. I updated the USB speed to 90 and that seemed to mostly correct the issues with that. However, regardless what I do, the settings don't seem to be saving regarding gain. 

Autoexposure works and seems to be following the settings. If I disable that and then manually set the exposure to a minimum setting (at night with the clock right on the RPi) the exposures seem to still auto adjust and the Gain settings that I applied manually in the gui and saved do not apply to the image. They are always 0 gain. 

I would also like to set the dark frame for the camera but with the issue of the autoexposures over-riding my settings, I can't get the frame to expose the same length as my longest exposure setting. I would think that should be the default setting or at least an option to select for that task.

Would anyone have any advice on these issues?

Frankastro85, 2020-06-07:
Hello,
You are alone...
Please read this topic :
[](https://github.com/thomasjacquin/allsky/issues/166)

EricClaeys, 2021-09-16:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.

Version 0.8 improved dark frame capture, so please try it.  If gain isn't working in version 0.8, turn on "Show gain" in the GUI and increase the Debug Level (an Advanced option) to 3 and look in /var/log/allsky.log to make sure it has the correct gain number.

closed 2021-09-16 by EricClaeys