AllskyTeam/allsky #195: ./capture process cannot find the ibtiff.so.4 library

wintersspace, 2020-06-08:
Hello

When I initiate the start command, I receive a response that the ./ capture process cannot find the ibtiff.so.4 library nor is it in any repository.

/home/pi/allsky $ ./allsky.sh preview
Starting allsky camera
./allsky.sh: línea 16: entr: orden no encontrada
ls: no se puede acceder a 'Whitehorse-full.jpg': No existe el fichero o el directorio
./capture: error while loading shared libraries: libtiff.so.4: cannot open shared object file: No such file or directory

Any solution?

thanks and regards

Luis

thomasjacquin, 2020-06-11:
Can you post your settings? I'm not sure why it's trying to use the TIFF library.

wintersspace, 2020-06-11:
Thanks for answering,
My configuration is a Raspberry PI 3 B + with Astroberry
https://www.astroberry.io/
Is it possible that Astroberry is not compatible with Allsky Camera?

Thank you
Luis

thomasjacquin, 2020-06-11:
I wouldn't say it's not compatible but there might be a conflict somewhere. Can you try without the `preview` argument and see if it's creating an image in `/home/pi/allsky`?

wintersspace, 2020-06-12:
Hello,

I am going to install a new astroberry image without installing additional software.
I will install AllSky again.

No images are generated. It is possible that there is a conflict with some program

wintersspace, 2020-06-13:
Hello,

I have mounted an SD card with an Astroberry image and installed Allsky.

At the time of installation has given this notice:

make -C sunwait-src
make[1]: se entra en el directorio '/home/pi/allsky/sunwait-src'
gcc -c -Wall  sunwait.cpp -o sunwait.o
sunwait.cpp: In function 'int main(int, char**)':
sunwait.cpp:664:62: warning: value computed is not used [-Wunused-value]
     if (arg[0] == '-' && arg[1] != '\0' && !isdigit(arg[1])) *arg++;
                                                              ^~~~~~
gcc -c -Wall  sunriset.cpp -o sunriset.o
sunriset.cpp: In function 'void sun_RA_dec(double, double*, double*, double*)':
sunriset.cpp:151:18: warning: variable 'zs' set but not used [-Wunused-but-set-variable]
   double xs, ys, zs;
                  ^~
gcc -c -Wall  print.cpp -o print.o
gcc sunwait.o sunriset.o print.o sunwait.h sunriset.h print.h -o sunwait -lm -lstdc++
make[1]: se sale del directorio '/home/pi/allsky/sunwait-src'
cp sunwait-src/sunwait .

* Sunwait installation
* Using the camera without root access

* Autostart script

* Configure log rotation

* Copy camera settings files
chown: usuario inválido: «pi:pi»
Created symlink /etc/systemd/system/multi-user.target.wants/allsky.service -> /lib/systemd/system/allsky.service.

And when starting Allsky this error occurs and does not copy images:

astroberry@astroberry:/home/pi/allsky $ sudo ./allsky.sh preview
[sudo] password for astroberry: 
can't get debug descriptor: Resource temporarily unavailable
Starting allsky camera...

 ******************************************
 *** Allsky Camera Software v0.6 | 2019 ***
 ******************************************

Capture images of the sky with a Raspberry Pi and an ASI Camera

Add -h or -help for available options 

Author: Thomas Jacquin - <jacquin.thomas@gmail.com>

Contributors:
-Knut Olav Klo
-Daniel Johnsen
-Yang and Sam from ZWO
-Robert Wagner
-Michael J. Kidd - <linuxkidd@gmail.com>

Listing Attached Cameras:
- 0 ZWO ASI120MC

ZWO ASI120MC Information:
- Resolution:1280x960
- Pixel Size: 3.8Î¼m
- Supported Bin: 1 2 
- Color Camera: bayer pattern:GR
- SDK version 1, 15, 0430
- Initialise Camera OK
- Sensor temperature:23.700000

Capture Settings: 
 Image Type: ASI_IMG_RGB24
 Resolution: 1280x960 
 Quality: 95 
 Exposure: 10000ms
 Max Exposure: 20000ms
 Auto Exposure: 1
 Gain: 50
 Max Gain: 200
 Cooler Enabled: 0
 Target Temperature: 0C
 Auto Gain: 0
 Brightness: 50
 Gamma: 50
 WB Red: 53
 WB Blue: 90
 Binning: 1
 Delay: 10ms
 Daytime Delay: 5000ms
 USB Speed: 40
 Text Overlay: text
 Text Position: 15px left, 30px top
 Font Name:  0
 Font Color: 255 , 255, 255
 Small Font Color: 0 , 0, 255
 Font Line Type: 16
 Font Size: 0.7
 Font Line: 1
 Outline Font : 0
 Flip Image: 0
 Filename: image.jpg
 Latitude: 60.7N
 Longitude: 135.05W
 Sun Elevation: -6
 Preview: 1
 Time: 1
 Darkframe: 0
 Show Details: 1

Saving auto exposed images every 10 ms

Press Ctrl+C to stop

OpenCV Error: Bad flag (parameter or structure field) (Unrecognized or unsupported array type) in cvGetMat, file /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/array.cpp, line 2501
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/array.cpp:2501: error: (-206) Unrecognized or unsupported array type in function cvGetMat

./allsky.sh: línea 32:  2128 Abortado                ./capture $ARGUMENTS
astroberry@astroberry:/home/pi/allsky $ 

thanks and regards

Luis

wintersspace, 2020-06-13:
Hello

I update the post.

I have managed to generate the image liveview-image.jpg and refresh normally.

I think it's a pi folder rights issue.

When starting with ./allsky.sh preview it doesn't work.

However with sudo ./allsky.sh it starts normally.

Lastly, ask if there is any way to simulate a meteor capture, to verify that the capture works before installing the camera.

thanks for everything

Luis

wintersspace, 2020-06-14:
Hello, I am unlucky because now I am having problems again.
I can not access the graphical interface, I have tried in these ways:

  http: //allsky.local
  http: // localhost
 
or directly by entering the IP of the Raspberry host in the browser

I have installed allsky-portal several times but I cannot start it

the result is 403 forbidden

There seems to be an error ..........

astroberry@allsky:~ $ systemctl status lighttpd

* lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2020-06-13 18:17:43 CEST; 1min 28s ago
  Process: 4649 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=0/SUCCESS)
  Process: 4654 ExecStart=/usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf (code=exited, status=255/EXCEPTION)
 Main PID: 4654 (code=exited, status=255/EXCEPTION)

jun 13 18:17:43 allsky systemd[1]: lighttpd.service: Service RestartSec=100ms expired, scheduling restart.
jun 13 18:17:43 allsky systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 5.
jun 13 18:17:43 allsky systemd[1]: Stopped Lighttpd Daemon.
jun 13 18:17:43 allsky systemd[1]: lighttpd.service: Start request repeated too quickly.
jun 13 18:17:43 allsky systemd[1]: lighttpd.service: Failed with result 'exit-code'.
jun 13 18:17:43 allsky systemd[1]: Failed to start Lighttpd Daemon.

some help?

thanks and regards

Luis

thomasjacquin, 2020-06-18:
Hi Luis, this is probably a folder permission again. Who's the owner of /var/www/html in Astroberry?

wintersspace, 2020-06-21:
Hi Thomas

Thanks for answering.

I think there is some kind of incompatibility with Astroberry, I can connect without problem through https://astroberry.local to the Raspberry but not through http://allsky.local nor access from the Raspberry locally using http://localhost

For some user with sufficient knowledge, it would not be a problem, but even if I change permissions and make changes, I cannot connect.

My intention is to capture meteors, and for that it works perfectly.

I use the allsky-website interface to view the images locally and I have a question.

Should the image be refreshed automatically every so often?

Now it only does this by pressing the F5 key in the browser

I would like to have the best configuration for capturing meteors, mine is this but it is sure to be improved so that I do not lose any event or meteor entry in the camera.

   "width":"0",
   "height":"0",
   "exposure":"5000",
   "maxexposure":"20000",
   "autoexposure":"0",
   "gain":"50",
   "maxgain":"200",
   "autogain":"1",
   "coolerEnabled":"0",
   "targetTemp":"0",
   "gamma":"50",
   "brightness":"50",
   "wbr":"53",
   "wbb":"90",
   "bin":"1",
   "delay":"10",
   "daytimeDelay": "5000",
   "type":"1",
   "quality":"95",
   "usb":"40",
   "filename":"image.jpg",
   "flip":"0",
   "text":"text",
   "textx":"15",
   "texty":"30",
   "fontname":"0",
   "fontcolor":"255 255 255",

If anyone has any more optimized settings I'd like to try it out, I'd appreciate it.

Otherwise Allsky works very well.

Thanks and regards

CleverJay, 2020-12-27:
I attempted to install Allsky onto Astroberry as well. I believe that there is a hostname conflict, even though I changed the PiName with`sudo gui/install.sh astroberry`.  This was a couple weeks ago and I seem to remember the error message was related to the web server. I ended up just reverting to Allsky and Astroberry running on their own separate Pi devices. Which is the lazy but more expensive solution.

EricClaeys, 2021-09-16:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-16 by EricClaeys