AllskyTeam/allsky #203: Add support for the raspberry HQ cam

ediazcomellas, 2020-06-28:
Hi,

Raspberry has recently released a new "HQ" cam with interchangeable lenses. It is not as capable as ASI alternatives, but it is also way cheaper.  It works using the standard raspberry pi commands (raspistill &co)

Is it possible to integrate this camera to the soft? I can send you one to support the development. 

Best regards

Dentonknifeworks, 2020-06-28:
https://github.com/RobboRob/allsky?fbclid=IwAR3rpb4dWYmZlEwOPT-lF2_Ow3yaRo0kC8ciBGWmcr8MJEeEis9AZ9hkLb0

Its been done. Check this link it works.

ediazcomellas, 2020-06-28:
Thanks! installing right now.

universe241981, 2020-06-30:
Is this also usable on Pi camera v1 & v2 ?

DarKOrange75, 2020-07-09:
> https://github.com/RobboRob/allsky?fbclid=IwAR3rpb4dWYmZlEwOPT-lF2_Ow3yaRo0kC8ciBGWmcr8MJEeEis9AZ9hkLb0
> 
> Its been done. Check this link it works.

I can't make it work like the original script from Thomas. Auto Exposure/Auto Gain are not working at night with the Raspberry Pi HQ camera. I have to set Exposure/gain manually hence the image is totally blank during the day.

Dentonknifeworks, 2020-07-11:
I have also found the Auto Exposure/Auto Gain are not working at night.
But they do during the day. The Time overlay was not implemented from what I read on the other link. I to miss that information.
It may be better to post on the other link and not on this one.

Dentonknifeworks, 2020-07-11:
I have been trying to pull the time and date from the file saved on of the resized image saved on the WEB site then print it just above the photo. Should be able to do this in the index.html file just have not found the code yet. This would be better then nothing.

Dentonknifeworks, 2020-07-12:
I was able to expand on this for a date/time display on the web page.
One issue is that you have to do a CTL+F5 for it to update this loads all of the site and updates. Now looking for code to do the command when the resized image is loaded.
The version I added to my page is in the index.html file on my site at [www.dentonknifeworks.com](url) you can see how it works as of now.
For testing I saved this as getdate.html you also have to do a CTL+F5 to get the current date. If anyone has a better way to do this would love to see it.
<html>
<head>
<title>Last Modified</title>
<script type="text/javascript">
function getLastMod(){
var myFrm = document.getElementById('myIframe');
var lastModif = new Date(myFrm.contentWindow.document.lastModified);
document.getElementById('LastModified').innerHTML = "Photo Date/time " + lastModif.toLocaleString();
}
</script>
</head>
<body>
<span id="LastModified"></span>

<iframe id="myIframe" onload="getLastMod()" src="image-resize.jpg" style="display:none;">
</iframe>

</body>
</html>

munnecke, 2020-07-16:
I have been running the RPi fork, but don't see any other place for commenting on it.  Hope this is the best forum.

I am running the lastest version from https://github.com/RobboRob/allsky doing 30 second exposures, but it is only capturing an image every 4 minutes.  Am I setting something wrong?  I would like to get as close to continuous coverage of the night as possible.

[image]

DarKOrange75, 2020-07-17:
> I have also found the Auto Exposure/Auto Gain are not working at night.
> But they do during the day. The Time overlay was not implemented from what I read on the other link. I to miss that information.
> It may be better to post on the other link and not on this one.

It may be an issue with the raspberry software. If you try to manually take a picture at night with raspistill and AE activated, the resulting picture is way too dark. This software is using raspistill to take the pictures.

DarKOrange75, 2020-07-17:
> I have been running the RPi fork, but don't see any other place for commenting on it. Hope this is the best forum.
>
> I am running the lastest version from https://github.com/RobboRob/allsky doing 30 [...]

I don't know if it will fix your issue but I've noticed on my setup that spaces in the image title are freezing the software.

munnecke, 2020-07-17:
I would never expect auto exposure to work for astrophotographs...  

I have the exposure dialed in OK, the problem that it only gives me one image every 4 minutes.

Raspistill does a lot of work "under the hood" - it has a timelapse option captures a sequence of images.

I'm happy to move the discussion to another forum if that is more appropriate.  What is the url of the other link you are referring to?

DarKOrange75, 2020-07-18:
> I would never expect auto exposure to work for astrophotographs...
>
> I have the exposure dialed in OK, the problem that it only gives me one image every 4 minutes.
>
> Raspistill does a lot of work [...]

You can post your question here: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=275801

DarKOrange75, 2020-07-18:
Maybe the following could be an interesting solution:

 https://github.com/thomasjacquin/allsky/issues/206

EricClaeys, 2021-09-16:
This issue is from an older release of the AllSky software.  Version 0.8 supports adding the time, gain, and exposure to the image.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed. 
Thanks.

closed 2021-09-16 by EricClaeys