AllskyTeam/allsky #205: Changing from RGB to RAW

WirthmU, 2020-07-15:
I would want to try out RAW. RGB is working fine. I am using a zwo ASI385MC camera.

I change the following in settings.json:

"type":"2",
"quality":"5",
"filename":"image.png",

Unfortunately I do not see any more daytime pictures on the website as I usually do if settings are setup for RGB.

Is there anything else to change?

minnican, 2020-07-21:
I just tried this and changing to RAW - to be honest I'd forgotten that I could use RAW - and found that using type 2 just produced a completely white image with the sensor temp/exposure/gain stamp only. Changing to type 0 produced a B&W presumably RAW image as expected and continued to save during the day. I haven't changed the filename, I left it as image.jpg and I restarted the service. All seems to be OK. I'm not sure this is at all helpful but I'd look at the filename checking that it has been changed to image.png everywhere.

WirthmU, 2020-07-21:
You are right - leaving the name at image.jpg did the trick. And as you said type 2 does not work and type 1 give a B&W picture on the website.
So here is my corrected settings.json

"type":"0",
"quality":"100",
"usb":"100",
"filename":"image.jpg",

WirthmU, 2020-07-24:
What would be the reason why the RAW picture has a pattern whereas the RGB looks better? It's the same for the timelapse. The live RAW picture does not show these squared patterns.

RAW:

RGB

minnican, 2020-07-24:
I had the same issue with the raw showing a checkerboard pattern. The image file sizes were bigger for the RAW than the JPEG, as you would expect, however the image quality was very poor. I changed back to RGB as I couldn't work it out.

Alex-developer, 2020-08-06:
I found an issue with the code that draws the text when using Raw16.

cv::putText(img, text, cvPoint(x, y), fontname, fontsize, cvScalar(fontcolor[0], fontcolor[1], fontcolor[2]), linewidth, linetype);

The RAw16 data is a mono image (CV_16UC1 ) so doesn't expect a BGR value for the colour. I have fixed this in my local, using a 178MM by hardcoding the value to white for now.

cv::putText(img, text, cvPoint(x, y), fontname, fontsize, 0xffff, linewidth, linetype);

If using Raw8 then the vale sholf be oxff

I will have a play around at the weekend and do a proper pull request

thomasjacquin, 2020-08-06:
Ah! Thanks Alex, that's a great find. Thanks for working on that.

Alex-developer, 2020-08-06:
> Ah! Thanks Alex, that's a great find. Thanks for working on that.

Problem is whats the best way to sort it? I thought about adding another field to the UI for the 'colour' to use when using Raw8 or 16 given that an BGR value makes no sense and trying to use the same field for both would not be easy.

Does that make sense Thomas? If so I'll make the changes and do a PR at the weekend

thomasjacquin, 2020-08-06:
@Alex-developer Since RAW images are grayscale, then maybe you can convert the colour value to a 16bit or 8bit one. That way, you keep the UI the same. I think OpenCV has functions to do the conversion 24bits RGB to grayscale.

Alex-developer, 2020-08-06:
> @Alex-developer Since RAW images are grayscale, then maybe you can convert the colour value to a 16bit or 8bit one. That way, you keep the UI the same. I think OpenCV has functions to do the conversion 24bits RGB to grayscale.

Thanks Thomas i'll take a look :-)

Alex-developer, 2020-08-12:
@thomasjacquin  Now have greyscale text on RAW16 captures - Just need to do the other text such as Sensor temp and will do a PR

[image]

EricClaeys, 2021-08-01:
Are these changes in 0.7 ?
Somewhere I saw a comment that RAW16 only works with .png files.  Is this correct?
I also saw in ZWO's documentation that RAW8 is for mono cameras and Y8 is for color cameras.  In the enhancements to allsky a few of us are working on, capture.cpp now supports Y8, but I've never tested to see what Y8 looks like on a color camera compared to RAW8 on a similar mono camera, e.g., ASI178MC vs ASI178MM.
Eric

closed 2021-09-16 by EricClaeys