AllskyTeam/allsky #206: Feature Request - INDI support

pterodyne, 2020-07-17:
If you aren't aware of INDI, please check here:

https://www.indilib.org/what-is-indi/features.html

It would make this project camera agnostic if the capture routine were switched to use indi instead of the ASI API.  I realize this is a considerable re-tooling, and I don't have the skills to do it.  But it seems that tons of feature requests are for supporting various camera manufacturers.  This is the whole reason for INDI.

pterodyne, 2020-07-17:
This is the list of INDI camera drivers currently supported:

https://www.indilib.org/individuals/devices/cameras.html

so all these cameras would be directly supported if this were implemented.

pterodyne, 2020-07-17:
Ill let it go, but one more thought.  The system with the camera wouldn't even have to be the system running the allsky software because INDI connects to devices over TCPIP.  Also things like filter wheels could be implemented fairly trivially

pterodyne, 2020-07-20:
While Id love to see INDI support in this project, Unfortunately the INDI driver for Raspberry Pi cameras probably also uses raspistill.  It's possible of course that it does something at a lower level.

Just researched it and it's not clear.  The question could be asked on the indi forums.

jkoenig72, 2020-08-23:
I had same idea, will look into it,

pk825, 2021-04-08:
Any news about extend the software to INDI?

jkoenig72, 2021-04-09:
That would be wonderful, right?

It's not that easy, I have started but could not find enough time. I used
python so far, but finding/implementing a good bayering function for color
cameras is challenging. I have a simple script in my repro - but very basic.

Regards
Joerg

pk825, 2021-04-09:
> That would be wonderful, right? It's not that easy, I have started but could not find enough time. I used python so far, but finding/implementing a good bayering function for color cameras is challenging. I have a simple script in my repro - but very basic. Regards Joerg
> […](#)

Thank you for info. I could use the program with Starlight Oculus and I have only found https://github.com/zemogle/pyOculus. I'll check soon your repo, maybe could help me in order to work with my cam

EricClaeys, 2021-09-16:
@thomasjacquin, @ckuethe,
any interest?  Seems like it would be nice but likely a huge change.

ckuethe, 2021-09-16:
Yes, I have interest in this. Aside from ZWO cameras, I also have some QHY, Touptek, and Celestron (v4l2)