AllskyTeam/allsky #207: FTP upload not working

MichaelSkycam, 2020-07-19:
Hello,
I searched the Issues here but could not find a solution.

Everything is working fine with my Allsky-Installation, the Images/Videos/Strartrails are saved without problems, but no files are uploaded to my Webspace.
I installed ftp on my Rasbian, all upload folders are chmod 777.
My ftp settings in Allsky are:

#!/bin/bash
# FTP/SFTP settings
PROTOCOL='ftp'
USER='xxxxxxxxxx'
PASSWORD='xxxxxxxxx'
HOST='xxxxxxxxxx.de'
IMGDIR='/allsky/images/'
MP4DIR='/allsky/videos/'
KEOGRAM_DIR='/allsky/keograms/'
STARTRAILS_DIR='/allsky/startrails/'

So you see, I did no changes and are using the Original-Form of the ftp-settings.
But I can`t figure out why ftp-Upload is not working. Is there a way to see if/what the Errors are ?

It would be nice to get this last Problem solved. Then my Skycam will be ready to go Online.
If you Thomas could help me, I will for sure donate some beers.

Thanks and best regards
Michael

Dentonknifeworks, 2020-07-19:
You can test your FTP login from a FTP client and be sure that your credentials are correct. This could be done from a PC. When you connect with a FTP client be sure that the dir structure is correct I had to change mine to XXXXX/allsky as my WEB root was under another DIR. Hope that helps.

MichaelSkycam, 2020-07-19:
Thanks for your fast response Thomas.

File-Upload with Filezilla works fine, the directory structure is correct. It`s just www.domain.de/allsky/images for example.
I tried now in Allsky FTP-Settings putting www before the Host and deleting the Backslash after the Directory Paths, but it`s still not working.

Dentonknifeworks, 2020-07-19:
First I am not Thomas LOL
This is what mine looks like
#!/bin/bash

# FTP/SFTP settings
PROTOCOL='ftp'
USER=‘xxxxxxx’  
PASSWORD=‘xxxxxxxxxxx’
HOST='dentonknifeworks.com'
IMGDIR='/home/allsky'
MP4DIR='/home/allsky/videos/'
KEOGRAM_DIR='/home/allsky/keograms/'
STARTRAILS_DIR='/home/allsky/startrails/'

I am trying to remember what I had to do with the PI to get it to work I think I installed FTP Also  there a SSH setting in the Pi Confguration that I turned on. I see now that I had installed LFTP not sure if that makes a difference. 
Bet Thomas has a fix if nothing I have given you works.
Brad

Dentonknifeworks, 2020-07-19:
Also have a look at the allsky.log file in /var/log.  You should see the issue in it.

MichaelSkycam, 2020-07-20:
Thanks Brad,
In /var/log I see only hundrets of the same fatal Error:

cd: fatal error: Certificate verification: certificate common name doesn't match requested host name »xxxxxxxxxx.de« 
put: image-resize.jpg: fatal error Certificate verification: certificate common name doesn't match requested host name »xxxxxxxxxxx.de« 

What could be the Problem ?
I don`t use SSL on my Webspace.

closed 2020-07-28 by MichaelSkycam