AllskyTeam/allsky #208: Getting both text overlay and time displayed

Lorenzi70, 2020-07-20:
Hi, I am trying to modify Capture.CPP to get overlay text AND ate-time displayed at the same time.
I see that there is a little piece of coding to un-comment that should serve my purpose:

//cvText(pRgb, ImgText, iTextX, iTextY+(iYOffset/bin), fontsize, linewidth, linetype[linenumber], fontname[fontnumber], fontcolor, Image_type, outlinefont);

Uncommenting it it works, however as the text overlay contains spaces it only shows the first part of the string and does not display the part after the space. Is there any way to fix it?
Thanks
Marco

EricClaeys, 2021-08-01:
If you are decent with programming, you can uses, for example, underscores for spaces, then have capture.cpp replace the underscores with spaces.  For example, enter "this_is_an_example".
A few of us are working on enhancements to allsky that, among other things, allows the text overlay AND time to appear, and allows spaces in all text fields.
We're close to submitting to Thomas for review.
Eric

EricClaeys, 2021-09-16:
This has been fixed in version 0.8, which now allows spaces in the "Text" option.  Further, the format of the time can be changed.

closed 2021-09-16 by EricClaeys