AllskyTeam/allsky #210: Adding frame number in timelapse

ediazcomellas, 2020-07-23:
Hi,

I find it useful to get the frame number in the timelapse. That way it is way easier to find interesting images for meteors or other events. 

To do this, I've just added this parameter to the ffmpeg command:

-vf "drawtext=fontfile=Arial.ttf: text='%{frame_num}': start_number=1: x=(w-tw)/10: y=h-(2*lh): fontcolor=black: fontsize=10: box=1: boxcolor=white: boxborderw=5"

The info was obtained from https://stackoverflow.com/questions/15364861/frame-number-overlay-with-ffmpeg

I would love to see this integrated as an option. If it is not, then this issue will be a good reference for those that want to change the timelapse script manually.

Dentonknifeworks, 2020-07-26:
> Hi,
>
> I find it useful to get the frame number in the timelapse. That way it is way easier to find interesting images for meteors or other events.
>
> To do this, I've just added this parameter to the [...]

Can you provide the full FFMPEG command? I am not sure at what part of the command to add this to. Front, Middle, etc.

EricClaeys, 2021-09-16:
A future version of Allsky will include a TIMELAPSE_PARAMETERS variable in config.sh that can be used to add parameters to the ffmpeg command.

closed 2021-09-16 by EricClaeys