AllskyTeam/allsky #211: FTP Upload again

MichaelSkycam, 2020-07-28:
Hello,

I really love the Allsky-Software, but the FTP Feature is a pain in the ass. I tried now for weeks to get it working, without success.
**Nothing is getting uploaded.**

My FTP settings are:
PROTOCOL='sftp'
USER='xxxxxxxxx'
PASSWORD='xxxxxxxxx'
HOST='xxxxxxxxxx.de'
IMGDIR='/images/'
MP4DIR='/allsky/videos/'
KEOGRAM_DIR='/allsky/keograms/'
STARTRAILS_DIR='/allsky/startrails/'

I checked the Paths with Filezilla, they must be right. I tried a thousand different combinations without success.
Of course lftp is installed and SSH is activated on my Raspberry. Settings in Allsky are also fine.

The Main-Directory of my Webspace has only one Sub-Directory "allsky". Very simple. The images Directory is in the Main-Directory like the ftp-settings say.
I had to change to **sftp** because **ftp** caused a "_Certificate verification: certificate common name doesn't match requested host name_" Error.

**At the Moment there a two Problems:**

- Nothing is getting uploaded to my Webspace. Why, I have no more Ideas.

- If I use the terminal command "_sudo service allsky status_" and watch the logs in var/log/allsky I see this Error:
"_allsky allsky.sh[817]: can't get debug descriptor: Resource temporarily unavailable_" What does this Error mean ?

I would be very lucky to get some help here !
Thanks and best regards
Michael

Dentonknifeworks, 2020-07-28:
Ok I think your issue is on the server side let me tell you what I had to do to get mine working.
When I go to dentonknifeworks.com it is actually on the server dentonknifworks.com/home/ 
So in my FTP setting I have 
HOST='dentonknifeworks.com'
IMGDIR='/home/allsky'
MP4DIR='/home/allsky/videos/'
KEOGRAM_DIR='/home/allsky/keograms/'
STARTRAILS_DIR='/home/allsky/startrails/'
Now this works for me but for one thing when I go to dentonknifworks.com/home/alsky/ it loads.
But I just want to put out dentonknifeworks.com so in the \home dir I put a redirect to www.dentonknifeworks.com/allsky/
This is the index.html in my home dir.
<head>
  <meta http-equiv="refresh" content="0; URL=http://www.dentonknifeworks.com/allsky/" />
</head>
<body>
  <p>If you are not redirected in one seconds, <a href="https://www.dentonknifeworks.com/allsky/">click here</a>.</p>
</body>

You should be able to look at your WEB site host and see if you have a home directory.
I hope this makes some sense to you but as I say look at your host settings. My /home/ is transparent.  Have you tried /home/ like mine above? Good luck.

thomasjacquin, 2020-07-28:
Yes, I think your issue is with the Path variables. They are the absolute paths on your server. For example, my keogram path is `/lamp0/web/vhosts/www.thomasjacquin.com/htdocs/allsky/keograms`.
Use your favourite FTP software (FileZilla for example) to find the proper path.
[image]

MichaelSkycam, 2020-07-30:
Sorry Guys, but that doesn´t work.

Filezilla and Cyberduck only tells me that the Path to my Maindirectory is "/". Like I made it in the Ftp-Settings above.
I contacted my Webhoster and they told me the real FTP-Path for example to the images-Dir is:
**/www/htdocs/[ftpuser]/[myhomepage.de]/allsky/images**

So I tried this FTP-Settings:

PROTOCOL='sftp'
USER='xxxxxxx'
PASSWORD='xxxxxxxx'
HOST='myhomepage.de'
IMGDIR='/www/htdocs/[ftpuser]/[myhomepage.de]/images'
MP4DIR='/www/htdocs/[ftpuser]/[myhomepage.de]/allsky/videos'
KEOGRAM_DIR='/www/htdocs/[ftpuser]/[myhomepage.de]/allsky/keograms'
STARTRAILS_DIR='/www/htdocs/[ftpuser]/[myhomepage.de]/allsky/startrails'

**Nothing is getting uploaded with this settings.**
If I use _sudo service allsky status_ I don`t get Error Messages.
But in var/log/ I can still find some of this Errors:
_Jul 30 15:25:13 allsky allsky.sh[809]: can't get debug descriptor: Resource temporarily unavailable_

Best regards
Michael

ckuethe, 2020-07-31:
I have some ideas about refactoring the upload feature because I'm not using ftp at all. Instead I use rsync or scp with key auth.

MichaelSkycam, 2020-08-01:
Hello ckuethe,

Thanks for your reply.
Unfortunatly my coding skills are a bit weak, so I´ve no Idea huw to implement rsync or scp.
I´m fine with ftp if it would be working.....

Best regrads
Michael

ckuethe, 2020-08-01:
Rsync is awesome. My upload command is roughly a 1-liner: `rsync -e 'ssh -i /home/allsky/id_allsky' --partial --inplace /home/allsky/images/artifacts allsky@www:` 

but that's only because I control the whole infrastructure so I can use SSH key auth, and I can set the home directory for the allsky user on the www server to be the right path. But I think this will be an easy change with the new config file editing UI. Then it's basically "uncomment and edit a few variables in this section to use scp to upload to your webserver" or the equivalents for sftp, rsync, ...

MichaelSkycam, 2020-08-02:
Thanks, that sounds interesting.
But at the Moment I would be happy to get the FTP-Problem solved. Actually I´m quite lost.
I hope I can get a few more Replies to solve this Riddle.

MichaelSkycam, 2020-08-10:
I fixed it.
In fact it was a Problem with the Paths, although I tried a Million Combinations.
If someone uses the German Webhoster All-Inkl.com the correct ftp-settings are:

PROTOCOL='ftp'
USER='[All-Inkl-User like w66oo66]'
PASSWORD='All-Inkl FTP-Passwort'
HOST='[All-inkl-User].kasserver.com'
IMGDIR='/allsky/images'
MP4DIR='/allsky/videos'
KEOGRAM_DIR='/allsky/keograms'
STARTRAILS_DIR='/allsky/startrails'

closed 2021-09-16 by EricClaeys