AllskyTeam/allsky #215: Raspberry Pi 4 - Does this board work with AllSky

spacemanbobby, 2020-08-06:
I remember when the 4 boards came out that there was some issues with it not working quite right.... 

Now that they have been out a while.. Has this been updated and does the AllSky work with them after installing all the normal files... I have copies of my older paths for FTP etc.. that I can just place after.

Dentonknifeworks, 2020-08-07:
I have been running mine on a PI4 for 2 months now.

Alex-developer, 2020-08-12:
Works 100%

spacemanbobby, 2020-08-12:
I'm getting serious image corruption during the day.. Nothing at night but
crazy scrambled images...

thomasjacquin, 2020-08-12:
The Pi 4 works. Your issue is probably caused by something else such as slow wireless connection or frame rate too high during the day (increase daytime delay).

spacemanbobby, 2020-08-12:
Ok... didn't have this with the Pi3 version. But I'll check things out this
weekend. I have the daytime delay set to 30 seconds. I am going to turn off
the stretch option. Also...

curitsreno, 2020-08-13:
Don't know if this helps or not. I have an ASI120mc and tried it on both a
PI3 and 4. Same results, or at least similar results. Good one second then
weird scrambled picture, I would get bars across the screen or the screen
would break up into 2 to 4 parts and be reassembled the wrong way. And what
fixes it for me is changing resolution from 0, 0 on the height, width
settings to 1200 x 950. Once I back down the resolution to those specific
numbers it never fails again. I went to the PI 4 hoping it would fix the
issue, thinking that usb2 was causing the issues, but it didn't fix it.

I think there are some serious issues with the canned drivers for this
camera and raspberry PI boards. Wish the hardware manufacturer (ASI120mc)
would step in and release proper drivers.

spacemanbobby, 2020-08-13:
Ok... In my config I have an ASI185 and I uncommented that line out to use
that in my Pi4 Board.. I looked at my old config in my Pi3 board and saw
that even though I was using the asi185mc I was using the config for the
the other asi camera.. So I changed it in the pi4 board to the original.
Now I am getting images with no garbled lines in it.

spacemanbobby, 2020-08-14:
I have noticed that the Dark frame isn't taking or isn't being applied. My
images have all sorts of heat noise on them. I have the dark frame option
turned on at night and it just doesn't seem to be applied to any of the
images. I am taking jpg images and quality is set to 100. I take the dark
around 9pm my time and then switch it back to off.
Here is a link to the days video for August 13th. The early part was before
I changed the config to use the default asi camera... thus the bad images
in the beginning.
http://spacemanbob.com/allsksy/videos/allsky-20200813.mp4

This is the path... from the configs section
# Path to the dark frame for hot pixels subtraction. Can be jpg or png.
DARK_FRAME="dark.jpg"

DarKOrange75, 2020-08-17:
> Ok... In my config I have an ASI185 and I uncommented that line out to use that in my Pi4 Board.. I looked at my old config in my Pi3 board and saw that even though I was using the asi185mc I was [...]

Hi, which line and in which file did you uncomment ?

spacemanbobby, 2020-08-18:
I had uncommented out the old config where you would use the asi185
settings... I have since just wiped the buster pi software and started
from scratch. New Pi software image... and new AllSky download. Looks like
things are all working.. I think it was a combo of issues at first.. Old
buster install and an older allsky download... combined with updating
both...

Now I am just running my asi185 on the default installed software and
everything seems to be working.

EricClaeys, 2021-09-16:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-16 by EricClaeys