AllskyTeam/allsky #216: Moon in the sky - no (minimal) startrails?!

si6pack, 2020-08-10:
Hi all,
I am running 2 allsky cams in parallel - everything is working fine, however, whenever there is the moon in the sky, startrails fail at all (just a random image of the night) or only short trails...any thoughts?
[image]
[image]
Thank you in advance,
si6pack

minnican, 2020-08-15:
This will be because the brightness of the images used to create the startrails exceeds the "brightness" threshold that you have set in the config.sh and they are therefore discarded. If you haven't changed the brightness from 0.1 in the config.sh I would increase it.  I'm currently using 0.2 in my setup.

si6pack, 2020-08-16:
You are absolutely right, that's it! Thank you very much!

I decided to get the job done manually with a higher threshold (.2). It works, but throws this error:
OpenCV Error: Unspecified error (could not find a writer for the specified extension) in imwrite_, file /build/opencv-U1UwfN/opencv-2.4.9.1+dfsg1/modules/highgui/src/loadsave.cpp, line 275
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-U1UwfN/opencv-2.4.9.1+dfsg1/modules/highgui/src/loadsave.cpp:275: error: (-2) could not find a writer for the specified extension in function imwrite_

This is in a way strange, since startrails always work when used during allsky service...
Regards,
Christian

thomasjacquin, 2020-08-16:
Can you paste your manual command here? Did you forget to add the `jpg` argument?

si6pack, 2020-08-16:
Here is the command I issued:

./startrails ./images/20200805/ jpg 0.2 ./images/20200805/startrails.jpg

Furthermore, the same error occurs, if I issue a manual keogram command…

I am in the „allsky“ Directory…

I am sorry for the inconvenience.

Kindest regards,
Christian

Von: Thomas Jacquin
Gesendet: Sonntag, 16. August 2020 09:37
An: thomasjacquin/allsky
Cc: si6pack; Author
Betreff: Re: [thomasjacquin/allsky] Moon in the sky - no (minimal)startrails?! (#216)

Can you paste your manual command here? Did you forget to add the jpg argument?

EricClaeys, 2021-09-16:
Is this still an issue with the newer version 0.8 code?  If not, would you please close the issue?  Thanks.

closed 2021-10-05 by ckuethe