AllskyTeam/allsky #217: Dark Frame not created

MichaelSkycam, 2020-08-10:
Hello,

For some reason I can`t create a Darkframe.
I followed exactly the Instructions for manual and GUI creating a Darkframe, but no Dark.png showes up in the Allsky Directory.

Is the creation of a Darkframe only possible at Nighttime as I`ve read in an older Issue ?

Best regards
Michael

si6pack, 2020-08-12:
I observe the same issue - no Dark.png after following exactely the manual instructions...Pi4, latest allsky software installed

Kindest regards,
Christian

MichaelSkycam, 2020-08-14:
In fact it seems to be Bug in the Allsky Software.

It works if you execute this script manually.
/home/pi/allsky/scripts/saveDarkframe.sh

Just cover your Camera Lens, wait some seconds until the next exposure starts, execute saveDarkframe.sh and uncover your lens again.

Best regards
Michael

si6pack, 2020-08-14:
Thank you very much for coming back!
Kindest regards from Gauting nearby Munich,
Christian

Von: MichaelSkycam
Gesendet: Freitag, 14. August 2020 16:47
An: thomasjacquin/allsky
Cc: si6pack; Comment
Betreff: Re: [thomasjacquin/allsky] Dark Frame not created (#217)

In fact it seems to be Bug in the Allsky Software.
It works if you execute this script manually.
/home/pi/allsky/scripts/saveDarkframe.sh
Just cover your Camera Lens, wait some seconds until the next exposure starts, execute saveDarkframe.sh and uncover your lens again.
Best regards
Michael

EricClaeys, 2021-08-01:
The 0.7 version of capture.cpp saves the dark frame as "image.jpg" (or whatever you have it set to be), not "Dark.png", which are then moved to the ~/allsky/darks folder with names like "39.jpg" for a dark taken when the sensor was 39 degrees.
Eric

EricClaeys, 2021-09-16:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Version 0.8 puts dark frames in ~/allsky/darks.

closed 2021-09-16 by EricClaeys