AllskyTeam/allsky #219: Google Chrome quesiton

maserowik, 2020-08-12:
Not sure if anyone else has this issue but I have a new one .
When using Chrome (84.0.4141.125) I have to do a control f5 to clear out my google cache to have the website update.
When using Internet Explore the site load up with a fresh copy.

mike

Dentonknifeworks, 2020-08-13:
Version 84.0.4147.125 (Official Build) (64-bit) is working for me. Sometimes fixing issues with chrome is just reinstall chrome.

Dentonknifeworks, 2020-08-13:
After reading this again it is true that control f5 will clear the cache. Are you having a issue with the image not updating automatically?

maserowik, 2020-08-13:
> After reading this again it is true that control f5 will clear the cache. Are you having a issue with the image not updating automatically?

That is correct, when I make a change on the allsky site I have to do a ctrl f5 for it the change to show.  Using IE or Firefox I do not have to do anything it happens automatically

I may have something set in chrome to cache.

Dentonknifeworks, 2020-08-13:
When you say changes? Changes in the Index.html or ? If it is changes in the WEB site yes you need to Control f5 to dump the cache in Crome for the site changes.

maserowik, 2020-08-13:
When I changed a setting in the config.js.  I have to clear cache on google chrome.

Dentonknifeworks, 2020-08-13:
Yes thats not the fault of Allsky but Chrome. I just got in the habit of doing the Ctrl F5 when ever I am working on the WEB page. There maybe a way to tell chrome not to cache your site. Also I seem to recall HTML code in the index.html that set the if the cache is expired or stored.

maserowik, 2020-08-13:
OK I did not think it was an ALLSKY issue, since it worked fine in IE and FireFox.

closed 2020-08-13 by maserowik