AllskyTeam/allsky #220: RAW16 Startrails and Keogram are all white

Alex-developer, 2020-08-12:
Having moved everything to RAW16, fixing the text colour issues i have just noticed that both the StarTrails and Keogram images generated by the end of night script are all white.

The source images are fine, there are no white images in the source and adjusting the BRIGHTNESS_THRESHOLD makes no difference.

@thomasjacquin Assuming this is a bug and not something I have missed in the manual, which is entirely possible, I can have a look at it if you like and try and fix it.

[image]

Alex-developer, 2020-08-16:
Had a quick look at this but not got very far as I don't know OpenCV that well (yet :-) )

I have a feeling its related to this line in startrails.cpp

accumulated = cv::max(accumulated, image);

@thomasjacquin any pointers you have would be appreciated.

thomasjacquin, 2020-08-17:
Thanks for the heads up @Alex-developer, I'll capture a few RAW 16 images tonight and try running the startrails code. I'm not an OpenCV expert at all but I'll try to debug this.

thomasjacquin, 2020-08-17:
@alex I haven't seen this bug yet. I am running the command manually and tested it at various threshold.

`pi@allsky:~/allsky $ ./startrails images/20200816/ png 0.4 images/20200816/startrails/startrails-20200816.png`

[image]

Alex-developer, 2020-08-17:
@thomasjacquin Which camera did you use? I am using a 178mm. I was hoping you were a cv expert ;-) I have tried it on both a 120mc and the 178mm with the same results

I'll do a complete clean install and continue to debug this end and see whats going on. 

Thanks

Alex

thomasjacquin, 2020-08-17:
I’m using the ASI 224mc and I did a full sd card re-install 2 days ago.
What’s the summary message when you run the startrails manually?

Alex-developer, 2020-08-17:
@thomasjacquin 

[606/614] image-20200817042804.png 0.360642
[607/614] image-20200817042845.png 0.369002
[608/614] image-20200817042925.png 0.377013
[609/614] image-20200817043006.png 0.385336
[610/614] image-20200817043047.png 0.394337
[611/614] image-20200817043127.png 0.404178
[612/614] image-20200817043208.png 0.414337
[613/614] image-20200817043249.png 0.423274
[614/614] image-20200817043330.png 0.431867
Minimum: 0.0885454 maximum: 0.431867 mean: 0.190034 median: 0.18749

Thats the end of last nights images for the startrails, the threshold is set t 0.4 but still a white image

EricClaeys, 2021-09-16:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

EricClaeys, 2021-09-16:
By the way, in version 0.8, 16-bit files are only supported with .png output, since with .jpg all-white images are often produced.

closed 2021-09-16 by EricClaeys