AllskyTeam/allsky #222: ASI120MC-S / rPI 2 "splitted" capture

iDiabDev, 2020-08-13:
After capturing OK for a while -> "splitted" captures occour
[image]
How to avoid/solve this?

maserowik, 2020-08-13:
Some folks are able to get the asi120MC-s working other are not.  I gave up on the 120 and went to a 224.

 

 <https://user-images.githubusercontent.com/49480463/90097926-5d829480-dd37-11ea-8eba-1eddcac50f6b.png> 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

RobertG73, 2020-08-13:
> 
> […](#)
> -- This email has been checked for viruses by Avast antivirus software. https://www.avast.com/antivirus

I think... i will do the same thing.

Alex-developer, 2020-08-14:
I had horrible problems with the 120MC so gave up and moved to an ASI178MM camera with much better results.

jkoenig72, 2020-08-23:
I had same issue, moved to raspi 3 and that solves it for me.

Respi84, 2020-08-29:
Hey! I set up everything yesterday, worked great that night but now I’m having that split screen problem, asi 120 PI 3B...
Is the only solution not using a 120?

EricClaeys, 2021-09-16:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  I'm not sure there's anything in the software that will fix this, although you can try adjusting the USB bandwidth.  Thanks.

closed 2021-09-16 by EricClaeys