AllskyTeam/allsky #226: RPI4 and HQcam

WirthmU, 2020-08-18:
I just tried the latest version (0.7) with HQcam support.

It seems that allksky does not recognize the HQcam. I am running buster.

```
pi@meteotux:~/allsky $ sudo service allsky status
● allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Tue 2020-08-18 09:02:57 CEST; 33s ago
  Process: 750 ExecStart=/home/pi/allsky/allsky.sh (code=killed, signal=TERM)
 Main PID: 750 (code=killed, signal=TERM)

Aug 18 09:02:57 meteotux allsky.sh[750]:   -h, --help
Aug 18 09:02:57 meteotux allsky.sh[750]:       Show usage and help
Aug 18 09:02:57 meteotux allsky.sh[750]: ZWO Camera not found. Exiting.
Aug 18 09:02:57 meteotux sudo[769]:       pi : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl stop allsky
Aug 18 09:02:57 meteotux sudo[769]: pam_unix(sudo:session): session opened for user root by (uid=0)
Aug 18 09:02:57 meteotux sudo[769]: pam_unix(sudo:session): session closed for user root
Aug 18 09:02:57 meteotux systemd[1]: Stopping All Sky Camera...
Aug 18 09:02:57 meteotux systemd[1]: allsky.service: Main process exited, code=killed, status=15/TERM
Aug 18 09:02:57 meteotux systemd[1]: allsky.service: Succeeded.
Aug 18 09:02:57 meteotux systemd[1]: Stopped All Sky Camera.
```

WirthmU, 2020-08-18:
Got it - I have not noticed that I had to select RPiHQ in the config.sh file
```
# Choose between ZWO or RPiHQ
CAMERA="RPiHQ"
```
I do get pictures now - however with that artefact lines that have been described before:
[image]

thomasjacquin, 2020-08-18:
Regarding the artefact lines, you'd have to ask Rob (https://github.com/RobboRob/allsky). He wrote the RPiHQ implementation. I just merged his code into the main repo. I noticed that the lines disappear when you toggle the `show details` option off.

Dentonknifeworks, 2020-08-18:
WirthmU this links tells all about the artifact lines

https://github.com/RobboRob/allsky/issues/10

WirthmU, 2020-08-19:
With the `show details` option off the lines are gone, indeed. I hope then that raspistill will be corrected for that.

closed 2020-08-19 by WirthmU