AllskyTeam/allsky #227: Web GUI not for HQpiCam

Dentonknifeworks, 2020-08-18:
The Web GUI is not the same as the one that Rob had created.
The issues I see is things like auto exposure is not a option for the HQpiCam
The red and blue won't take a decimal point only whole numbers I do use 1.8 for a example.
The gain for the HPpiCam max is 16.
For someone that would install this for there HQpiCam I think would struggle badly with this.
Now I hope you just tell me Hi stupid turn on this and you will have the GUI for th HQpiCam LOL.
Thanks For you work!
Brad

thomasjacquin, 2020-08-18:
Hi Brad,

Thanks for the feedback, I haven't filtered the camera options yet. I merged the code from Rob yesterday and made sure it was functional enough to capture images. I will refine the camera settings page to display the appropriate settings soon.
Regarding the decimal point, that's a bug. I'll fix it.

Dentonknifeworks, 2020-08-18:
Cool Yes I have it capturing images. And I sure like the script editing makes it so much easier to make adjustments.
Thanks

thomasjacquin, 2020-08-19:
@Dentonknifeworksn The latest code should take care of that. I added a separate `camera_options_RPiHQ.json` file specific to the RPiHQ. The appropriate file is loaded based on what camera you have selected in `config.js`

Dentonknifeworks, 2020-08-19:
@thomasJacquin. Looks good only issue I see is the default values in cam settings on first load are for the ZWO this produced a black screen on Live Preview. I had the last values printed out so quick fix as I could see they where wrong for the HQPi cam. If I see anything else after a run tonight I will let you know. Oh I did have to add source $ALLSKY_HOME/scripts/ftp-settings.sh to saveimageDay.sh and postData.sh for them to connect to the web site. This was running them from the Term as a FTP test for me. I seen some one else comment on it.

EricClaeys, 2021-09-16:
This issue is from an older release of the AllSky software and appears to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-16 by EricClaeys