AllskyTeam/allsky #228: Flickering sky images on AllSky

spacemanbobby, 2020-08-18:
Getting Flickering images during the capture of images during the day and night. Not garbled, but just brightness over exposures every so often during the images that are taken. 

Gain and Exposure are turned to "NO" so auto exposure and Gain don't change. 33 second captures. 
http://spacemanbob.com/allsksy/videos/allsky-20200817.mp4

Using Pi4 Buster current updated version and fresh install of allsky from scratch. 
ASI185MC camera. 
Didn't have this type of thing when operating on Pi3 version before I upgraded.

thomasjacquin, 2020-08-18:
During the day, auto-exposure is always used no matter what setting you're using. This is to achieve a properly exposed image during the day.
The flickering is very likely caused by a bug in the auto-exposure calculation in the ZWO library. It seems to affect only a few models and is somewhat recent. The ZWO library was updated in March. This could be where the issue comes from.

Try using an older version by replacing /lib/armv7/libASICamer2.a by this one https://github.com/thomasjacquin/allsky/raw/bd03b5b09f5bac4eaac75a93ddf1c79b489a3b71/lib/armv7/libASICamera2.a

daleghent, 2020-08-19:
FWIW, I see the same effect on a 290MC.

spacemanbobby, 2020-08-19:
I have replaced that file to my board... I'll report my findings when the
camera gets a few hours of images saved to the SD Card.

spacemanbobby, 2020-08-19:
I replaced the libASICamera2.a file from the github link.

Still the same result. Images are going over exposed still.
Want me to try something else.

thomasjacquin, 2020-08-20:
Can you try an even earlier one? This one is from last year.
https://github.com/thomasjacquin/allsky/raw/4c218fe4ca41d9c0a03055b8b8c9dbecc4f22c6e/lib/armv7/libASICamera2.a

spacemanbobby, 2020-08-20:
How about the one that was used on Pi3.
The one you gave me is still doing the over exposure.. I stopped the allsky
service, changed out the file and started it back up.. dang it...

spacemanbobby, 2020-08-20:
Ok.. I have just for the heck of it.. Done some things...
I have removed the libASICamera2 files from all the ARM directories and x64
and x86 dir... Nothing changed.. still image flickering... I then removed
the ASI Camera2.h file.. restarted everything.. camera works like before.
Then I removed the ASIconfig.xml from the .zwo folder. Restarted.. Same..
Where is it pulling the asi camera from if I took it out of the other
areas.
hmm

thomasjacquin, 2020-08-20:
Try recompiling the software after you make changes.
`rm capture`
`make capture`

spacemanbobby, 2020-08-20:
Did the make capture and started allsky back up.. Still getting the
exposure issue. Only happens during the day though

EricClaeys, 2021-08-01:
Like Thomas said, it's a ZWO bug.  They are aware of it and it's on their "to do" list.
In the meantime, a few of us are working on enhancements to allsky that include, among others, a workaround for ZWO's bug.  It's not perfect but it's MUCH better.  We're just about ready to post for Thomas to review.
Eric

EricClaeys, 2021-09-16:
We pretty much fixed this in version 0.8 by using our own auto-exposure algorithm.  If you haven't done so already, please upgrade to version 0.8.

closed 2021-09-16 by EricClaeys