AllskyTeam/allsky #229: QHY support

daleghent, 2020-08-19:
Hi,

I'm in the process of building an all-sky camera and plan to use either an ASI290MC or a QHY5-III 462C that is currently on order. Of course, there is no QHY-specific `capture` command. I created the native QHY driver in [NINA](https://nighttime-imaging.eu) and am willing to create the necessary `catpure` command that is specific for QHY cameras and uses their Linux/ARM ports of their SDK.

Looking at the `capture` command, it is specific to ZWO and there is a separate source file for the RPi HQ camera as well. Would you be open to the idea of creating a single-source `capture` command where hardware support is modularized, so that one command can be complied and the user selects which camera type they wish to use via the existing camera settings?

thomasjacquin, 2020-08-19:
Hi Dale,

I started working on something like this last night. The idea is to have a `capture_QHY.cpp` file. I started a branch here https://github.com/thomasjacquin/allsky/tree/qhy-camera

The camera selection happens in config.sh `CAMERA="QHY"`. It's not functional yet but I don't think it would take too long. I'll have to clean up that branch and remove all the unused files at some point... still experimenting!

I'll try to get the capture working tonight but feel free to try it on your side.

daleghent, 2020-08-19:
Oh wow! How fortuitous. If you'd like, I can look the QHY code over, test it with my cameras, and submit any changes or suggestions.

thomasjacquin, 2020-08-19:
Oh yeah for sure! Go ahead. Last night I managed to get a compilation with no failure but when I start it manually with `allsky.sh` It wasn't connecting to the camera. Could be because there were 2 others cams plugged to the Pi.
Anyways, I've had the CPP file capture an image at some point in the past but the RG and B channels were side by side on the image... it's definitely a work in progress. Let me know how it goes.

daleghent, 2020-08-19:
Ok. My two QHY cameras are a QHY600M and QHY183M, so I don't have a color sensor on hand at the moment but I should get the 462C later this week, which will get me a OSC to work with. 

Is your plan to have a distinct capture command for each camera, including all non-camera-specific supporting logic? I was thinking that the camera-specific bits could be on their own but share all the common code so as to minimize duplication

thomasjacquin, 2020-08-19:
I agree, It would be nice to avoid duplication and have a code base more INDI like. Although I think the `capture_RPiHQ` is quite different since it's using `raspistill` and doesn't support the same camera settings.
ZWO and QHY could probably share a lot of code in common.

daleghent, 2020-08-19:
Well, at the end of the day, each camera must provide a specific thing - an image bitmap or other image-y thing. That's really what we need, right? There are is always going to be differences amongst SDKs and even the hardware that will make it impossible to 1:1 map every command line option to `capture` to a function on the camera... but that's alright. Depending on the camera, some options will be no-ops, or take slightly different parameters. Each camera driver can advertise what it's capable of and we would match the logic accordingly, as best as we can. I'll see if I can work out a PoC, though my C++ is a little rusty, I should be able to manage.

thomasjacquin, 2020-08-19:
That's right, all we need is a binary that produces an image.
I created separate camera_options_XXX.json files specific to the camera model. These get used in the GUI to display the appropriate options. It would be too hard to keep one file that applies to all brands (different options, different default values, different ranges).
e.g. https://github.com/thomasjacquin/allsky-portal/blob/master/camera_options_RPiHQ.json

daleghent, 2020-09-04:
I've been finding some time to work on this branch locally. A few things:

1. I talked to QHY and they are currently working on actual auto-exposure in the SDK. The QHY_3A_AUTO* controls currently don't actually do anything, so I guess we look for it in a future SDK release. This really only affects daytime shooting anyway since nighttime shots are a pinned exposure length.

2. Regarding SDKs in general, I've worked on creating a SDK download and installation script that allows the user to choose which SDK(s) they want to install onto their system. The various allsky utilities are compiled using them, with the binaries being linked to the SDK's shared objects. This allows the user to upgrade the SDK in the future without having to recompile the utilities as well (modulo any API/ABI-breaking changes in the SDKs themselves, with is a very rare event)

3. I have `capture_QHY` properly debayering color images now and exposing correctly. When time allows, I'm going to suss out the options mapping. As I'm doing this, I'm looking at ways to turn the ZWO, QHY, and RPI camera stuff into modules that are usable by a single `capture` executable.

thomasjacquin, 2020-09-08:
Hi Dale, these are great news. It will definitely be a nice addition to the family of cameras supported by the software. I haven't spent too much work on the QHY code so there shouldn't be any issues merging your code in. I have just been fixing bug reports from users after releasing v0.7. There are still minor quirks but it should be stable soon and it'll be a good code base to use when merging your QHY work.

Shand359, 2020-10-22:
I am just wondering if this is a thing now, with the QHY camera support?

I only have a QHY5P-II Coloe camera but willing to at least help testing....

Shand359, 2020-10-31:
Has there been any progress in utilising the QHY cameras for the allsky setup...??

claudioarena, 2021-01-25:
I am interested as well. Is the QHY branch working at the moment?

I also have sort of a weird question - I have an old QHY5 that I have resurrected (it was producing black images) by desoldering the main image sensor and soldering in a new one. However, I soldered a colour one instead of the original B&W.

Working great but I imagine using this branch would produced a B&W image. Where in the code should I look if I want to debayer the image to display it as colour one? or is that done by the SDK for colour images?

Thanks!

WirthmU, 2021-08-09:
Is the QHY branch still usable? 

I would try to use it with a QHY5L-II Colour camera.

Thanks

EricClaeys, 2021-09-17:
@daleghent,  any update on this?  FYI, the version 0.8 capture.cpp has a lot of changes from 0.7, and a newer version I'm working on has even more changes.

@ckuethe, see the posts above - perhaps you can work with Dale on QHY support?

closed 2021-10-16 by EricClaeys