AllskyTeam/allsky #230: FTP and RPiHQ

WirthmU, 2020-08-19:
While FTP is working on a different AllSky system with an ASI camera it does not with RPiHQ.

If I try to upload manually I get:
```
pi@allsky:~/allsky/scripts $  ./saveImageDay.sh
cat: temperature.txt: No such file or directory
Resizing
Uploading

pi@allsky:~/allsky/scripts $ open: Not connected
put: image-resize.jpg: Not connected
^C
```
If I use lftp on the command line I get:
```
pi@allsky:~/allsky/images/20200818 $ lftp -e "debug; set ssl:verify-certificate no; set ftp:ssl-allow no; put /home/pi/allsky/images/20200818/startrails.jpg -o /web/Bern/allsky/startrails/startrails.jpg; bye" -u user,password XX.XXX.XXX.XXX
---- Connecting to XX.XXX.XXX.XXX (XX.XXX.XXX.XXX) port 21
<--- 220 IoT_Bern FTP server ready.                                         
---> FEAT
<--- 211- Extensions supported:                                                   
<---  AUTH TLS
<---  PBSZ
<---  PROT
<---  CCC
<---  SIZE
<---  MDTM
<---  REST STREAM
<---  MFMT
<---  TVFS
<---  MLST
<---  MLSD
<---  UTF8
<--- 211 End.
---> OPTS UTF8 ON
<--- 200 OK, UTF-8 enabled                                                  
---> USER admin
<--- 331 Password required for user.
---> PASS password
<--- 230 User admin logged in.                                                    
---> PWD
<--- 257 "/" is current directory.                                                    
---> TYPE I
<--- 200 Type set to I.                                                               
---> PASV
<--- 227 Entering Passive Mode (XX,XXX,XXX,XXX,XXX,XX)
---- Connecting data socket to (XX.XXX.XXX.XXX) port 55837
---- Data connection established
---> ALLO 6628662
<--- 202 ALLO command ignored.
---> STOR web/Bern/allsky/startrails/startrails.jpg
<--- 150 Opening BINARY mode data connection for 'web/Bern/allsky/startrails/startrails.jpg'.
---- Closing data socket                                                                           
<--- 226 Transfer complete.                                                                                           
---> MFMT 20200819063907 web/Bern/allsky/startrails/startrails.jpg
<--- 213 Modify=20200819063907; web/Bern/allsky/startrails/startrails.jpg                                             
6628662 bytes transferred in 5 seconds (1.17 MiB/s)
---> QUIT
<--- 221 Goodbye. You uploaded 6473.30 KB and downloaded 0 bytes.
---- Closing control socket
```

Is it the `cat: temperature.txt: No such file or directory` that causes the problem? Does the RPiHQ generate such a file at all?

thomasjacquin, 2020-08-19:
The temperature issue is not really an issue. It's used to create and load temperature dependant darks.
The command you pasted says that you uploaded 6.4MB. That seems like the proper size of an image. Are you sure it's not somewhere else on your web host? Usually the startrails images have a date in the filename, not sure why yours doesn't.

WirthmU, 2020-08-19:
The `startrails.jpg` was just an example file I used because I wanted to check if the directory on my server is accessible and if lftp would actually work with my settings.

The problem is that the `./saveImageDay.sh` does not connect to my server with the same settings and no current live picture is uploaded anywhere on the server.

WirthmU, 2020-08-19:
I figured it out - it is a permission issue which I need to correct:

WirthmU, 2020-08-19:
After I added the line
`source $ALLSKY_HOME/scripts/ftp-settings.sh`
to the top of the saveImageDay.sh script it worked.

closed 2020-08-25 by WirthmU