AllskyTeam/allsky #231: Permissions and subdirectories

jlnorthrup, 2020-08-19:
I am using a RP4, GUI, and HQ camera and noticed during the initial run last night that while it saved the images in the folder titled with yesterday’s date, it had permissions errors in the creation of sub folders for the keogram, Star trails, and Timelapse. I was able to manually create a keogram by using CLI commands and saving it outside of the ./images/20200818/ folder.  When looking at the permissions for /images/ they are set to Only Owner and that seems to be limited by something.

WirthmU, 2020-08-20:
Same here - images are saved in their respective directories but neither keograms nor startrails are generated by the daily routine. Both can be produced by hand using the ./keograms or ./startrails commands in their respective subdirectories.

In my settings the permissions are ok, though.

jlnorthrup, 2020-08-20:
A bit more information. It appears folders that are created by the GUI say the owner is Root verses current user.  
Example:
mkdir -p creates a directory for tonight and a sub directory thumbnails both owned by Root
Images are saved but also owned by root
endOfNight.sh runs unable to make additional subdirectories because this script isn’t seen as Root user.

thomasjacquin, 2020-08-20:
Thanks for the info! That's useful, I'll check where the issue is coming from.

thomasjacquin, 2020-08-20:
Can you take a look at `/lib/systemd/system/allsky.service` and make sure that there is a line saying `User=pi`. That's the user that is used to run the service and therefore all the scripts. When `.endOfNight.sh` is run, it's run by this user so `mkdir -p` should create directories owned by pi and not root.
If those directories are owned by root, that's because some command was manually run as root (aside from allsky.sh).

jlnorthrup, 2020-08-20:
That looks like the issue. In the allsky.service the user was blank and has now been changed and services shutdown and machine restarted.

thomasjacquin, 2020-08-25:
What user do you have in /lib/systemd/system/allsky.service?
There should be a line saying `User=pi`. That's the user that runs the service. That means the directories should be created by pi, not root.

jlnorthrup, 2020-08-25:
When that file is initially written the user showed as blank.  I use nano to edit it to say User=pi then then I did save restart before doing the chown to fix the ownership errors.  After installing the GUI install.sh, I checked allsky.service and it was still saying User=pi but the files and folders changed in the GUI install were back to root ownership. 
I’m currently running Linux 5.4.51-v71+ arm71 RaspOS

jlnorthrup, 2020-08-25:
This is really interesting. I deleted the allsky directory and fixed /lib/systemd/system/allsky.service to say User=pi
Then I re did the install as just ./install.sh without sudo.  It installed everything again but it wasn’t able to change that allsky.service back to blank.  I then tried to do the GUI/install.sh and it did the steps of the process but even after restart it didn’t want to start.  All the ownerships were correct but it wouldn’t open ./allsky.sh preview or GUI.
I deleted the directory again and repeated the verification that user was still pi.  It then did the install again with sudo’s, as soon as the install command hit the write allsky.service part of the install it changed the file (I assume to switch User=pi to User= )

thomasjacquin, 2020-08-25:
Try opening a terminal and type `sudo logname` it should return `pi`. This is the command used to get the username in `allsky.service`.

jlnorthrup, 2020-08-26:
When is do sudo logname the response is logname: no login name

thomasjacquin, 2020-08-26:
That's the issue.
I will try on the new Raspberry Pi OS released in August. Maybe things have changed.
Mine is `Linux allsky 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux`

thomasjacquin, 2020-08-26:
Looks like a known issue https://github.com/jochemin/raspnode/issues/5

jlnorthrup, 2020-08-27:
Made a little change in config.sh
In the last line changed the command to CAMERA_SETTINGS=“./settings_$CAMERA.json”

jlnorthrup, 2020-08-29:
As a work around for my machine - it appears that a few Canakit RPi’s will not populate the logname correctly. As correction I changed in the install.sh’s all times it uses logname variable to pi and it is installing correctly.

ebeyonder, 2020-10-16:
Hi, does this mean your method requires a reinstallation of allsky?

jlnorthrup, 2020-10-16:
> Hi, does this mean your method requires a reinstallation of allsky?
@ebeyonder 
Yes, I had to go in and make the manual changes to the install.sh before I did the initial install of Allsky.  Is your Pi  purchased from Canakit too?  For some reason they don't save that logname variable so we have to do it manually.

EricClaeys, 2021-09-17:
@jlnorthrup, if this has been resolved would you please close the issue?  Thanks.

closed 2021-09-17 by jlnorthrup