AllskyTeam/allsky #240: Camera White Balance

cplonski, 2020-08-23:
I'm using a ZWO ASI224MC, and it's basically impossible to achieve reasonable white balance.  With the default settings of 50 for gamma and brightness, the setting of wbr=53 and wbb=90 is way too red, and you can't bring red low and blue up high enough to produce white.  However, if I change gamma to 49 or to 51 the image is seriously green with a fine checkerboard pattern over the image, and no combination of wbr or wbb works to balance this out.  Effectively the image can be too red or too green depending on the gamma setting being 50 or some other number.

across-suffolk, 2020-08-29:
To reduce the green you should increase both the blue and red levels. It helps to think of this like a tripod, adjusting the green leg means pushing up the blue and red legs.

EricClaeys, 2021-09-17:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-17 by EricClaeys