AllskyTeam/allsky #244: Nighttime images black, no apparent way to lighten

cplonski, 2020-08-24:
Nighttime images black, no apparent way to lighten.  tried manual gain settings, exposure, etc.  This did seem to work properly before running the latest release.

Camera ZWO ASI224MC

thomasjacquin, 2020-08-25:
The software probably thinks it's daytime where you are. Did you set the RPi time properly (timezone setting) and your latitude and longitude in the camera settings?

cplonski, 2020-08-26:
I did double checked date and time, and I'm pretty sure I got lat and long right unless it's looking for a - minus sign in front of 80.74W:

   "latitude":"34.69N",
   "longitude":"80.74W",

thomasjacquin, 2020-08-26:
Are you running auto-exposure and auto-gain at the same time? These 2 interfere and sometimes produce black images. Also, let the software run for a few minutes if you're using auto-exposure.

cplonski, 2020-08-26:
no, just auto-exposure.  I set the gain to 135 for unity gain, although it does not appear to care which gain setting I have as long as the exposures are way too short

thomasjacquin, 2020-08-26:
Can you post your settings?

cplonski, 2020-08-28:
They are just the defaults with my own lat and long.  

I'd post the settings, but I don't have them anymore.  I got frustrated and installed a RPiHQ cam and all the ZWO driver woes vanished, and the color balance/clarity/auto-exposure on the RPiHQ blew away the ZWO.

maphilli14, 2020-09-04:
I'm having similar issues on my ZWO174MM, it works fine all day then when it goes to nighttime the gain goes from 0 to 50 and stops (max is 250) and exposure goes from 0.0x to 3.187s and stops (max is 4)

maphilli14, 2020-09-04:
status reads: "ZWO Camera not found. Exiting."

   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Thu 2020-09-03 21:38:37 EDT; 1h 15min ago
  Process: 11379 ExecStart=/home/pi/allsky/allsky.sh (code=killed, signal=TERM)
 Main PID: 11379 (code=killed, signal=TERM)

Sep 03 21:38:31 allsky allsky.sh[11379]: can't get debug descriptor: Resource temporarily unavailable
Sep 03 21:38:36 allsky allsky.sh[11379]: cannot read device status, Resource temporarily unavailable (11)
Sep 03 21:38:36 allsky allsky.sh[11379]: ZWO Camera not found. Exiting.
Sep 03 21:38:37 allsky sudo[11437]:       pi : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl stop all
Sep 03 21:38:37 allsky sudo[11437]: pam_unix(sudo:session): session opened for user root by (uid=0)
Sep 03 21:38:37 allsky systemd[1]: Stopping All Sky Camera...
Sep 03 21:38:37 allsky systemd[1]: allsky.service: Main process exited, code=killed, status=15/TERM
Sep 03 21:38:37 allsky sudo[11437]: pam_unix(sudo:session): session closed for user root
Sep 03 21:38:37 allsky systemd[1]: allsky.service: Succeeded.
Sep 03 21:38:37 allsky systemd[1]: Stopped All Sky Camera.
~

cat allsky/settings_ZWO.json
{
   "width":"1216",
   "height":"1216",
   "exposure":"1000",
   "maxexposure":"4000",
   "autoexposure":"1",
   "gain":"50",
   "maxgain":"500",
   "autogain":"0",
   "coolerEnabled":"0",
   "targetTemp":"0",
   "gamma":"50",
   "brightness":"50",
   "wbr":"53",
   "wbb":"90",
   "bin":"1",
   "delay":"10",
   "daytimeDelay": "5000",
   "type":"1",
   "quality":"95",
   "usb":"40",
   "filename":"image.jpg",
   "flip":"0",
   "text":"text",
   "textx":"15",
   "texty":"30",
   "fontname":"0",
   "fontcolor":"255 255 255",
   "smallfontcolor":"0 0 255",
   "fonttype":"0",
   "fontsize":"7",
   "fontline":"1",
   "outlinefont":"0",
   "latitude":"35.7N",
   "longitude":"78.0W",
   "angle":"-6",
   "time":"1",
   "darkframe":"0",
   "showDetails":"1"
}

maphilli14, 2020-09-04:
Sorry for the spam, update to os w/ reboot:

lines 1-11/11 (END)
● allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-09-03 23:03:06 EDT; 48s ago
 Main PID: 1372 (allsky.sh)
    Tasks: 6 (limit: 2065)
   CGroup: /system.slice/allsky.service
           ├─1372 /bin/bash /home/pi/allsky/allsky.sh
           └─1448 /home/pi/allsky/capture -angle -6 -autoexposure 1 -autogain 0 -bin 1 -brightness 50 -coolerEnabled 0 -darkframe 0 -daytimeDelay 5000 -delay 10 -exposure 1000 -filename image.jpg -flip 0 -fontcolor 255 255 25

Sep 03 23:03:06 allsky systemd[1]: Started All Sky Camera.
Sep 03 23:03:06 allsky allsky.sh[1372]: can't get debug descriptor: Resource temporarily unavailable
~
~

same results for exp and gain, even with one auto pick, either exp or gain...

EricClaeys, 2021-09-17:
This issue is from an older release of the AllSky software and appears to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.

closed 2021-09-17 by EricClaeys