AllskyTeam/allsky #245: Text overlay overlapping

DarKOrange75, 2020-08-24:
Hi @thomasjacquin,

Just a heads up that the text overlay is messed up if you increase font size.

A picture is better than a thousand words ;-)

[image]

Thank you

wramey7274, 2020-08-31:
Hi DarkOrange75

Exactly, and I have been trying to increase line-height some how. I am using a ZWO 178MC camera. I wonder if it is camera dependent?

thomasjacquin, 2020-08-31:
Look for all instances of `iYOffset += 30;` in capture.cpp and change it to something that works for you. Then run `make capture` in a terminal to recompile.

wramey7274, 2020-08-31:
Thank you very much for such a quick answer.

DarKOrange75, 2020-08-31:
Thank you Thomas!

wramey7274, 2020-09-02:
DarkOrange75: I found 4 occurrences of +=30 in the file capture.cpp. Did you find any more? What perimeter did you change +=30? I am too new at this, and any help will be greatly appreciated.

DarKOrange75, 2020-09-02:
> DarkOrange75: I found 4 occurrences of +=30 in the file capture.cpp. Did you find any more? What perimeter did you change +=30? I am too new at this, and any help will be greatly appreciated.

I found 4 occurrences too, set all occurrences to 50:

iYOffset += 50;

then I compiled and restarted Allsky service and no more overlapping for me (font size 18, font weight 2)

wramey7274, 2020-09-02:
Thank you DarkOrange75 for your quick reply. I followed your lead and all is well.

DarKOrange75, 2020-09-02:
You are welcome

EricClaeys, 2021-09-17:
Closing - This issue is from an older release of the AllSky software and has been resolved in the current version.  There is an option allowing users to set the line height.

closed 2021-09-17 by EricClaeys