AllskyTeam/allsky #246: endOfNight_RPiHQ.sh: not found

WirthmU, 2020-08-25:
My endOfNight files are not created. I get the following error message in allsky.log:

`sh: 1: scripts/endOfNight_RPiHQ.sh: not found`

DarKOrange75, 2020-08-25:
Are you sure you have the latest version? Fresh install yesterday and I don't have this text in the GUI, instead: "Range for a PNG image: 0-9. Range for a JPG image: 0-100"

WirthmU, 2020-08-26:
It is the RPiHQ-version of the administration panel - thus no PNG choice. So I guess it is the latest version.

Also, but this is more about esthetics and a total detail, one might be more strict about the use of upper- and lower-cases in titles and text:
`Red **b**alance / Red component for the **W**hite **B**alance, range 0 - 10`

thomasjacquin, 2020-08-26:
Try changing the capture_RPiHQ.cpp line 991 to the regular one: `system("scripts/endOfNight_RPiHQ.sh &");`
and recompile: `make capture_RPiHQ`.
This is a typo I missed.

EricClaeys, 2021-09-17:
This issue is from an older release of the AllSky software and I believe has been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  The newest version does not have endOfNight_RPiHQ.sh.

closed 2021-09-17 by EricClaeys