AllskyTeam/allsky #247: Web GUI not working any more

WirthmU, 2020-08-26:
With the latest changes the Web GUI does no longer work. Lighttp does not start:

```
pi@allsky:~ $ sudo systemctl status lighttpd.service
● lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-08-26 09:01:55 CEST; 17s ago
  Process: 1564 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=255/EXCEPTION)

Aug 26 09:01:55 allsky systemd[1]: lighttpd.service: Service RestartSec=100ms expired, scheduling restart.
Aug 26 09:01:55 allsky systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 5.
Aug 26 09:01:55 allsky systemd[1]: Stopped Lighttpd Daemon.
Aug 26 09:01:55 allsky systemd[1]: lighttpd.service: Start request repeated too quickly.
Aug 26 09:01:55 allsky systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Aug 26 09:01:55 allsky systemd[1]: Failed to start Lighttpd Daemon.
```

thomasjacquin, 2020-08-26:
Check the logs `/var/log/lighttpd/error.log` and `/var/log/allsky.log`

Respi84, 2020-08-26:
Hey, I'm still having problems to set up everything...I now get the Message ERR_CONNECTION_REFUSED when I try to access  my local IP http://192.168.178.92/ or http://192.168.178.92/public.php or http://allsky.local
can someone help me, the raspberry is totally new to my so I don't know where to look for permissions ect...

maserowik, 2020-08-27:
so my sd card died and I had to reload my Raspberry Pi.  I am unable to get the gui to load also.

/var/log/lighttp/error.log

2020-08-26 16:08:11: (server.c.1464) server started (lighttpd/1.4.53)
2020-08-26 16:08:26: (server.c.2059) server stopped by UID = 0 PID = 1

/var/log/allsky.log
Aug 26 16:03:34 allsky allsky.sh[499]: can't get debug descriptor: Resource temporarily unavailable
Aug 26 20:18:50 allsky allsky.sh[578]: can't get debug descriptor: Resource temporarily unavailable
Aug 26 20:26:06 allsky allsky.sh[581]: can't get debug descriptor: Resource temporarily unavailable
Aug 26 20:40:56 allsky allsky.sh[571]: can't get debug descriptor: Resource temporarily unavailable

jlnorthrup, 2020-08-27:
Same problem, when I tried to check the log. It is telling me the lighttpd folder is owned by www-data

thomasjacquin, 2020-08-27:
I just removed a line from lighttpd.conf that was added recently through a pull request. Try running `sudo gui/install.sh` again and that should fix it.

maserowik, 2020-08-27:
Just tried it and still not loading

pi@allsky:~ $  sudo systemctl status lighttpd.service
● lighttpd.service - Lighttpd Daemon
   Loaded: loaded (/lib/systemd/system/lighttpd.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2020-08-27 14:40:22 EDT; 1min 35s ago
  Process: 636 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=255/EXCEPTION)

Aug 27 14:40:22 allsky systemd[1]: lighttpd.service: Service RestartSec=100ms expired, scheduling restart.
Aug 27 14:40:22 allsky systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 5.
Aug 27 14:40:22 allsky systemd[1]: Stopped Lighttpd Daemon.
Aug 27 14:40:22 allsky systemd[1]: lighttpd.service: Start request repeated too quickly.
Aug 27 14:40:22 allsky systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Aug 27 14:40:22 allsky systemd[1]: Failed to start Lighttpd Daemon.

/var/log/lighttpd/error.log
2020-08-27 14:39:33: (server.c.1464) server started (lighttpd/1.4.53)
2020-08-27 14:39:49: (server.c.2059) server stopped by UID = 0 PID = 1

thomasjacquin, 2020-08-27:
Oh sorry, there's another step to do before running the installer: Open `gui/lighttpd.conf` and remove this line `mimetype.assign             = (".css" => "text/css", )`

across-suffolk, 2020-08-27:
Thank you Thomas, that worked for me.

maserowik, 2020-08-27:
Also works now for me.
Thanks!!!!

Respi84, 2020-08-27:
Hey @thomasjacquin 
Thanks man! Awesome! It works for me too! Awesome project, thanks for sharing and doing great support! I just bought you the next cup of coffee!;)

Greetings from Germany!

thomasjacquin, 2020-08-27:
Thanks @Respi84. It's much appreciated!

closed 2020-08-31 by WirthmU