AllskyTeam/allsky #250: Captures off RPiHQ are at odd delays

jlnorthrup, 2020-08-30:
In GUI camera Settings I have the exposure set for 10000 ms and delay set for 1000 ms however I’m only getting a picture every 1.54 minutes on average (low=1.52 min and high=1.55 min)

The previous night I had it set for exposure 30000 ms and delay of 10 ms and was getting images every 3.22 min.

jlnorthrup, 2020-08-31:
Info from last night’s tests:
From sunset to midnight I had the delay set to 500 ms via the GUI and had actual delays of 3.22 min. 
From midnight to sunrise I used to GUI to change the delay to 2000 ms and it was still doing a 3.22 min delay between pictures. 
The log file is showing the changes are written but it doesn’t seem to be changing the pacing. I’m now on the hunt through the code to see if I can find reference to 193 or 192 because when multiplied by 1000 would be close to the delays I’m seeing.

DarKOrange75, 2020-09-02:
Please post a screenshot of your parameters

jlnorthrup, 2020-09-02:
@DarKOrange75 Here is a screenshot of the settings. 

[image]

Here is a section of the night’s images
[image]

Based on my settings I should be seeing a new image about every 16 seconds.

DarKOrange75, 2020-09-02:
Settings look OK. What is your hardware setup? Are you running something elision your Pi?

jlnorthrup, 2020-09-02:
That is the weird thing, there isn’t any other heavy use applications running. I have VNC and ssh enabled so I don’t have to take the case apart. According to the GUI’s system monitor the CPU load is between 1-5% and it is only using 4% of the 4GB of ram. 
I was getting 3:22 interval with 10,000 ms exposures, last night’s goal was to try and get the focus right so I lengthened the exposure time.

jlnorthrup, 2020-09-02:
The system setup is RPi4 4GB with a PiHQ camera with 180 lens. It is powered by a PoE hat and uses WiFi for its network connection.  No FTP uploads, I just connect through VNC and grad any files that are interesting.

jlnorthrup, 2020-09-03:
@DarKOrange75 I did an experiment last night. I changed the night and day delay to 10 ms (just in case there is a section of code that is multiplying the values). It is now pretty consistent with an interval of 2 minutes and 17 seconds 
[image]
[image]

DarKOrange75, 2020-09-03:
With 45000ms exposure time and 10ms delay I have the Rpi taking a picture every 50 seconds more or less

DarKOrange75, 2020-09-05:
You are welcome. I have the exact same hardware as you so if you still have issues after reinstall the issue could also come from your SD card that is too slow. Good luck.

jlnorthrup, 2020-09-07:
I did the reinstall. @DarKOrange75 started with your settings. I thought to test the SD card writing by setting it for 24hr saving. During the day it snapped a picture every 30 sec during the day and as soon as it started calling the capture at night it jumped back to 3 minutes

DarKOrange75, 2020-09-08:
Have you also reinstalled the OS?

jlnorthrup, 2020-09-08:
Just did this morning
I didn’t install the GUI this time, and will just go in by VNC to pull images from it.

jlnorthrup, 2020-09-09:
[image]
 A whole new error.

thomasjacquin, 2020-09-09:
Try running `raspistill -o test.jpg` in a terminal to see if the error comes from allsky or from the camera.

jlnorthrup, 2020-09-09:
It was camera, in removing the Pi from it's waterproof tube the ribbon cable became partially removed.  I shut down the Pi and fixed the cable.  After restart, it went active and only showing the normal lsusb messages again.  
[image]
I am going to run without a gui for a couple of nights to see if that solves the issue with the delays.

jlnorthrup, 2020-09-12:
I've now got 3 successful nights done, star trails, Keogram, and time lapse.  I have it set to 60 sec exposure and 10 millisecond delay and the system is running without a problem.  I have not put the GUI interface back on and just access the machine by VNC.  I am wondering if there is a conflict with using the PoE Hat and Wifi that when the GUI gets in the picture they start conflicting.  As of now it is behaving and I'm getting images that look great on our planetarium and the learn-from-home students are getting a kick out of the time lapse movies.  
@thomasjacquin - I am fine with closing this if you think it is a error unique to my setup.

thomasjacquin, 2020-09-12:
@jlnorthrup I’m glad it’s working for you now. The GUI should not interfere with the capture program so I’m not too sure what happened. The only thing I can think of is that without the GUI, the settings file is located at `/home/pi/allsky` and when using the GUI, the settings file is at `/etc/raspap`

EricClaeys, 2021-09-17:
This issue is from an older release of the AllSky software and appears to have been resolved.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.
FYI, you can increase the Debug Level in the GUI (or "debuglevel" in the .json file) to see additional information in /var/log/allsky.log which may give an idea where the delay is coming from.

closed 2021-09-17 by EricClaeys