AllskyTeam/allsky #251: Settings for day/night when using RPiHQ

cplonski, 2020-08-31:
The settings for day/night when using RPiHQ need to be different, namely Auto-gain for day and manual gain for night.  How can I get this to change from day to night and vice-versa?

DarKOrange75, 2020-09-02:
During the day the settings are on "auto", you just need to care about the settings for the night capture.

cplonski, 2020-09-02:
Well, that's the theory, but when I switched to the RPiHQ from the ZWO camera, it looks like the Auto-exposure setting is gone, so I have to rely on Auto-gain which does not appear to change day and night.  My question is how can I set the auto-gain on for daytime and off for night when using the HQ camera.

DarKOrange75, 2020-09-02:
Rpi HQ implementation is really new and the switch from ZWO to Rpi HQ might not have been planned. IMO it would be faster to start from a new and fresh install

cplonski, 2020-09-02:
I will as soon as the code shows any newer updates from when I did a fresh install a few days ago.  As of now, I still have to switch auto gain on and off manually

kingjamez, 2020-09-28:
Anyone have a solution for this? I'm not able to get the RPiHQ camera to work both day and night. Only either or.

thomasjacquin, 2020-09-28:
ZWO cameras have a built-in auto-exposure feature. The RPiHQ camera does not provide this feature. It might be achievable through software by analyzing the previous image brightness. I haven't looked into it yet though.

closed 2021-09-17 by EricClaeys