AllskyTeam/allsky #255: Timelapse creation very very slow

emielkempen, 2020-09-15:
Hello,
I have build an AllSky camera and it is working fine, exept for one thing.

The timelapse is not being created.

If I run the timelapse.sh script by hand it stops after 43 frames.
I found in this thread: https://github.com/thomasjacquin/allsky/issues/141 increasing the swap file might help.

I increased the swap file to 1024mb and if I run the timelapse.sh script by hand the script continues, but very very slow and there is no timelapse created or uploaded.

I have a Raspberry Pi 3B and a ZWO ASI178MC camera

Am I doing something wrong?
Thanks in advance.

Kind regards,
Emiel Kempen.

dbainbridge, 2020-09-15:
I have same camera.  The issue is is that avconv/ffmpeg is running out of memory because that camera has too high of resolution for the amount of memory available on the Pi3 for ffmpeg to process it.  You'll need to scale the image in the timelapse.sh file on the ffmpeg line.  I had to add `-s 1548x1040` which cuts the resolution in half and the Pi3 can handle that.  I'm still using older version that used avconv instead of ffmpeg but the flag should be the same.

This will reduce the resolution of the video file but I still find it acceptable.  Increasing swap file is probably not good idea since you would be thrashing the sd card every morning.  Other option would be to get RPi 4 with 4 or 8 GB of memory.

emielkempen, 2020-09-16:
That worked! The creation of a timelapse is working now.
Thanks for the quick response.
Emiel.

svego, 2021-01-26:
I've made the scale option configurable by configs. That way you don't have to edit the scripts and will be able to update with git more easy. Now waiting for @thomasjacquin to accept the [new code](https://github.com/thomasjacquin/allsky/pull/315) ;-)

thomasjacquin, 2021-01-28:
@svego's scale option is now available.

EricClaeys, 2021-09-17:
Closing issue - it's been resolved.

closed 2021-09-17 by EricClaeys