AllskyTeam/allsky #256: RPiHQ Binning: Images are extremly darker than without Binning

pmneo, 2020-09-15:
I'm testing the binning functionality to reduce image size, but in this case the frames are extremly dark.

Can anybody confirm this?
[image]
[image]

CS
pmneo

SRadiant, 2020-09-16:
2 2028 × 1520 4:3 0.1-50 fps Full 2×2 binned
expose only 10 sec
3 4056 × 3040 4:3 0.005-10 fps Full None
expose up to 200 sec

pmneo, 2020-09-16:
Oh .. thanks for the clarification ...

So :D i will take a look if i can use your crop feature to resize the images after captures

Thanks

closed 2020-09-16 by pmneo