AllskyTeam/allsky #259: Cannot create regular file: 'Liveview-image.jpg' Permission denied

wglogowski, 2020-09-18:
Hi, first I know just enough about programming to enter command lines ... just letting you know my knowledge base.  I loaded up the program and everything looks like it is working when I run the program.  I set up Pixel to use the desktop environment and I get a live screen image. When I try to find my images they don't seem to be saved.  Looking at the instructions on the GitHub page I found that I can add a preview argument to show images the program is currently saving.  When I do that the terminal window shows the error 
Exposure value: 64 ms (the exposure value changes with each attempt)
20200918 10:47:57 Auto Exposure value 0ms
cp: cannot create regular file 'Liveview-image.jpg' Permission denied

Can someone help me?

Thank you, Walter

wglogowski, 2020-09-18:
I should add that I am using a ZWO 120MM USB 3.0 camera

stansley, 2020-09-18:
Walter - following your CN post on this too (dswtan). At first glance it sounds like a simple permissions issue, which rings a bell for me when I set my system up and I had enough Linux knowledge to fix it easily myself. To check this, can you go to the command line (not GUI), enter these commands, and copy the results here please?

`cd ~` -- just makes sure you are in the "home" directory, where the allsky installation probably is
`ls -l` -- show the output of this to us
`ls -l allsky` -- show the output of this to us

wglogowski, 2020-09-19:
Thank you so much.  It turned out that I did not read your GitHub documentation carefully enough.  I thought that the images would be saved in the images folder even during the day. I went back to carefully read your page and also someone on CN mentioned that they thought it only saves them after the sun is below 6 degrees. I tried it just now at night and all the images are being saved!  

Thank you so much for your reply 

Walter

thomasjacquin, 2020-09-19:
@wglogowski There is a option to enable 24hr capture in config.js. It's called `CAPTURE_24HR`. Set it to true in order to enable capture during both day and night.

wglogowski, 2020-09-19:
Thank you!

Walter

EricClaeys, 2021-09-17:
Closing the issue as the problem has been resolved.

closed 2021-09-17 by EricClaeys