AllskyTeam/allsky #260: Additional text

Alex-developer, 2020-09-19:
@thomasjacquin Thomas, I have been adding support for displaying additional text, mainly for my remote dew heater control. I have made some changes to the code and just wanted to check with you before creating a pr just to make sure you are happy for this to be in the core code.

The main changes are

1) Add some new options to specify a file containing the additional text to display
2) An an option to stop displaying the contents of the file if its not been updated for x seconds, this is just a safety check in case the program thats updating the file stops and the images won't have out of data data on them - Setting this to 0 disables the check
3) I also took the opportunity to add a setting that specifies the vertical spacing between lines - Just so that people don't have to change the code, this is issue 245

Thanks

Alex

[image]
[image]

thomasjacquin, 2020-09-22:
Hi @Alex-developer, this is a great idea. I think that'll help a lot of people who have hacked the CPP file in order to display various info. Makes it easier for them to update to new versions of the software.
Good call on the line height option!

You can make a PR, I'll review it, do a fresh install and I'll let you know how it goes!

Thanks

Alex-developer, 2020-10-03:
Thanks @thomasjacquin Just doing a test install then I'll create a PR

Thanks

Alex

Alex-developer, 2020-10-04:
Code merged so closing

closed 2020-10-04 by Alex-developer