AllskyTeam/allsky #261: Timelapse and Startrail missing

wglogowski, 2020-09-21:
In the Config.sh file
I have:
TIMELAPSE = true
KEOGRAM = true
STARTRAILS = true 

When I look at the Images tab in the GUI I have a full night's worth of images ...all good.   However when I click on Timelapse, Keogram and or Startrails there are no files.

I attempted to manual  create a timelapse using the code ./scripts/timelapse.sh 20200920 (last night's file) but I get an error :
failed to create symbolic link '/home/allsky/images/20200920/sequence/ then the running number of images I have taken : No such file or directory.  Before I entered the code I cd to allsky this was confirmed by the command line changing to pi@allsky:~/allsky $

thomasjacquin, 2020-09-22:
It looks like it's failing to create the `sequence` directory in the `20200920`. It's probably a permission error.
Here's where the error occurs https://github.com/thomasjacquin/allsky/blob/master/scripts/timelapse.sh#L20

wglogowski, 2020-09-22:
Do you have advice on how to fix the permission?  

Walter

thomasjacquin, 2020-09-22:
I would make sure that the allsky directory (and all subdirectories) is owned by user `pi`.
`cd ~`
`chown pi:pi -R allsky`

wglogowski, 2020-09-22:
I get the message  chown changing ownership of ‘all sky/xxxxxxx’ :Operation not permitted

thomasjacquin, 2020-09-22:
You need `sudo`.
`sudo chown pi:pi -R allsky`

wglogowski, 2020-09-22:
Should I see anything happen?  the command line just went back to pi@allsky:~ $

wglogowski, 2020-09-22:
It seems to be be working now, thank you so much!

Walter

meteorkees, 2020-12-20:
All looks good but the timelapse generated are only 48kb. I did a manual run. No errors occured. Timelapse still empty.
I did also the sudo chown pi:pi -R allsky
Did not help. Any tips or suggestions?
Thanks!

EricClaeys, 2021-09-17:
This issue is from an older release of the AllSky software.  If you think it might still be relevant, please test it with the newest version and submit a new issue if needed.  Thanks.
Failure to create timelapse videos is usually due to insufficient memory.  You can try increasing the swap space, or decreasing the size of the images - see config.sh for the necessary variables.

closed 2021-09-17 by EricClaeys