AllskyTeam/allsky #264: Keogram and Startrails not uploading - Video OK

LifeSVR, 2020-09-26:
There might be several issues but this is the main one. 

For about a week, my allsky is losing connection a lot. It keeps capturing pictures though. 
At the end of the night, it doesn't make the 2 result images and video. 

When manually generating (by generateForDay.sh) it does create all three (as I can see them in the admin panel). 
But it only uploads the video to my server. 

When manually using uploadforday it returns this:
`
pi@allsky:~/allsky/scripts $ ./uploadForDay.sh 20200922
Uploading Keogram

open: Not connected
put: /home/pi/allsky/images/20200922/keogram/keogram-20200922.jpg: Not connected

Uploading Startrails

open: Not connected
put: /home/pi/allsky/images/20200922/startrails/startrails-20200922.jpg: Not connected

Uploading Timelapse

open: Not connected
put: /home/pi/allsky/images/20200922/allsky-20200922.mp4: Not connected
`

Is there any way to resolve this?

LifeSVR, 2020-11-02:
Still having issues. The end of night ritual is not working. The keogram, startrails and movie are not generating nor uploading to my NAS. 

When manually creating them (generateforday script), it does generate keogram and startrails, but the pi keeps getting an error on saving the movie: 

> Output #0, mp4, to 'images/20201029/allsky-20201029.mp4':
  Metadata:
    encoder         : Lavf58.20.100
    Stream #0:0: Video: h264 (libx264) (avc1 / 0x31637661), yuv420p, 1304x976 [SAR 1:1 DAR 163:122], q=-1--1, 2000 kb/s, 15 fps, 15360 tbn, 15 tbc
    Metadata:
      encoder         : Lavc58.35.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/2000000 buffer size: 0 vbv_delay: -1
frame=    2 fps=0.3 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=    7 fps=0.8 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   12 fps=1.3 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   17 fps=1.8 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   23 fps=2.3 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   26 fps=2.5 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   31 fps=2.8 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   36 fps=3.1 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   39 fps=3.2 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   42 fps=3.3 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   44 fps=3.3 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   47 fps=2.6 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   ./timelapse.sh: line 35: 30718 Killed                  ffmpeg -y -f image2 -r $FPS -i images/$1/sequence/%04d.$EXTENSION -vcodec libx264 -b:v 2000k -pix_fmt yuv420p -movflags +faststart images/$1/allsky-$1.mp4

thomasjacquin, 2020-11-05:
What Raspberry Pi version and what camera do you have?

EricClaeys, 2021-09-17:
Closing - no reply from user in over 10 months.

closed 2021-09-17 by EricClaeys