AllskyTeam/allsky #272: Image resizing

Xplode85, 2020-10-12:
Setting image width/height only cropes the image leaving the upper left corner, is that how it's supposed to work or is it supposed to resize the image?
I'm using an ASI178MC so being able to resize images to save space would be great

thomasjacquin, 2020-10-14:
Yes, this is how the ZWO library works. In order to save space, you could use the binning option.

Alex-developer, 2020-10-24:
If you only after saving space and using a 1.5mm lens you could use the Crop option in config.sh to remove all but just the image circle from the camera? I use a 178MM with a 1.5mm lens and there is a huge amount of dead space in the raw image so I crop this out.

# Crop the captured image, used to improve the images when using a fisheye lens
CROP_IMAGE=false
CROP_WIDTH=640
CROP_HEIGHT=480
CROP_OFFSET_X=0
CROP_OFFSET_Y=0

Xplode85, 2020-10-24:
I use a 2.7mm lens so there is no dead space

EricClaeys, 2021-09-17:
Closing - if this is still an issue, please open a new issue with additional details.

closed 2021-09-17 by EricClaeys