AllskyTeam/allsky #276: First time I have encountered this. Startrail

BrightskyObservatory, 2020-10-15:
I know some people have has Startrail generation issues before, but this is a first for me. When the Startrail was generated, it would only show a single frame. I tried doing it manually but at the end of the sequence it came up with this..

[1901/1918] image-20201015053138.jpg 0.423138
[1902/1918] image-20201015053148.jpg 0.401691
[1903/1918] image-20201015053158.jpg 0.407979
[1904/1918] image-20201015053208.jpg 0.414395
[1905/1918] image-20201015053219.jpg 0.42356
[1906/1918] image-20201015053229.jpg 0.416399
[1907/1918] image-20201015053241.jpg 0.394272
[1908/1918] image-20201015053252.jpg 0.399479
[1909/1918] image-20201015053303.jpg 0.405132
[1910/1918] image-20201015053314.jpg 0.41069
[1911/1918] image-20201015053326.jpg 0.416436
[1912/1918] image-20201015053338.jpg 0.409551
[1913/1918] image-20201015053350.jpg 0.426701
[1914/1918] image-20201015053403.jpg 0.418081
[1915/1918] image-20201015053415.jpg 0.399209
[1916/1918] image-20201015053428.jpg 0.406756
[1917/1918] image-20201015053441.jpg 0.411668
[1918/1918] image-20201015053454.jpg 0.408957
Minimum: 0.204289 maximum: 0.901081 mean: 0.244897 median: 0.218033
No images below threshold, writing the minimum image only

What would an ideal setting be, I tried 0.15, 0.3, 0.5, 0.7 none seem to generate a trail image. It has worked previously but I had an issue and had to reinstall the Allsky on the Pi. 

http://brightskyobservatory.bsch.com.au/allsky

Andrew.

thomasjacquin, 2020-10-15:
You're right, this is strange. I'd say you should get a pretty good startrails image with a .3 threshold. Can you post the manual command?

BrightskyObservatory, 2020-10-18:
Ok, never mind, it is working again, I must have done something wrong as it's working fine now. Sorry for the long delay.

closed 2021-09-17 by EricClaeys