AllskyTeam/allsky #277: Part of image noisier than rest?

ebeyonder, 2020-10-15:
Hi, I am using an ASI178MC with the allsky software. Tonight is the first night I am running it and I notice a quadrant of my image being noiser than the rest. Please see image below:

[image]

I have tried powering off, resetting etc, but this still remains. Any ideas would be appreciated. Thanks!

thomasjacquin, 2020-10-15:
Did you by any chance take a dark frame and do you have dark frame subtraction set to true?

ebeyonder, 2020-10-15:
Yes indeed! I had changed cameras and didn't switch off dark frame subtraction. Thank you so much!! Problem solved.

closed 2020-10-15 by ebeyonder