AllskyTeam/allsky #278: Cannot delete images

ebeyonder, 2020-10-15:
Hi all - I cannot seem to delete images in the /images folder, and my card is fast running out of space.

I did a check of /lib/systemd/system/allsky.service, and it seems that there is no 'pi' after 'user='

When I did a sudo nano to insert user=pi, the camera stopped taking images. I removed the 'pi' and it worked again.

I also checked the permissions of the subfolders under /images, and change content is set to 'only owner'. The owner is 'root' and group is 'root'. 

I suppose if I cannot delete the images, the system is also unable to auto-delete them? 

I would be most grateful for any help to enable the deletion of the images. Thank you.

thomasjacquin, 2020-10-15:
In theory, you should have `user=pi` in allsky.service
Also, the allsky directory and subdirectories should be owned by user `pi`
`chown -R pi:pi ~/allsky`
Then the service will be run by user pi and will be able to save images to the directories owned by pi.

ebeyonder, 2020-10-16:
Thank you. Thomas. Unfortunately when I did the chown command, I got a slew of errors saying permission denied (please see screenshot below). Any further help appreciated - thanks!

sudo nano /lib/systemd/syst[image]

ebeyonder, 2020-10-16:
I read a similar thread on permissions, and it seems I have the same problem.

I did this --> Try opening a terminal and type sudo logname it should return pi. This is the command used to get the username in allsky.service.

And my username also returns as blank.

Right now having changed the user=pi (but changed back later), it seems endOfNight.sh isn't working anymore - my stratrails, timelapse etc all don't get uploaded to my website. 

Would the solution then be to:

1. Still have user=pi using the sudo nano command
2. Do this chown -R pi:pi ~/allsky - but it does not seem to work?
3. Do this too? --> As a work around for my machine - it appears that a few Canakit RPi’s will not populate the logname correctly. As correction I changed in the install.sh’s all times it uses logname variable to pi and it is installing correctly.

For step 3, does this actually mean I need to reinstall allsky on my machine? Is the logname variable only here:

chown -R `logname`:`logname` ../allsky 

Thank you so much!

thomasjacquin, 2020-10-16:
You just need sudo with the chown command since your allsky directory is owned by root:
`sudo chown -R pi:pi ~/allsky`

ebeyonder, 2020-10-16:
Works now - I can delete the images! Thank you so much, Thomas, for the allsky and also the quite help! 

I will check tomorrow morning to see if the keograms, timelapse and startrails all get uploaded.

ebeyonder, 2020-10-17:
Thank you again - allsky did create a startrail, but not the video - it created a small sized video (48 bytes), uploaded it to my website. There was no keogram. I will set up a separate thread on this. Thank you so much!

closed 2020-10-17 by ebeyonder