AllskyTeam/allsky #279: No sequence folder, timelapse created but 48 bytes

ebeyonder, 2020-10-17:
Hello - so I had some permissions problem with the folders and could not delete images, and after the sudo chown pi:pi -R allsky command, I was able to delete folders, etc. 

Last night, allsky successfully created a startrail image, which it uploaded to my FTP (incorrectly to the root folder, but that's an issue for another day). 

It also uploaded a timelapse video - this time correct to the videos subfolder. However, it is only 48 bytes large and unplayable. There was no keogram created.

I went to check the subfolder under images, and there was no sequence subfolder created for last night. However, before I did this sudo chown pi:pi -R, the system was able to generate timelapses, etc and upload them to my website. All the previous nights' images subfolders contained the 'sequence' subfolder.

I ran the timelapse.sh script, and it proceeded (I think) to create indeed the same timelapse video of 48 bytes. I took some screenshots below; it does seem to have referred to a 'sequence' folder, though I did not see that anywhere later on.

There is also no keogram created - which I suppose is also dependent on the sequence folder?

Any help would be much appreciated! Thank you.

[image]
[image]
[image]
[image]

ebeyonder, 2020-10-17:
I should add that, from the image below, it seems to have created the timelapse of 7 mins, but something seems to have gone wrong between this point and the uploading?

The timelapse video in the image subfolder for last night is also 48 bytes, so it doesn't seem to be a problem with the uploading process.

[image]

ebeyonder, 2020-10-19:
Still no luck, I am afraid - despite uploading a new timelapse.sh from another thread that reduces the images before the timelapse is created.

It is the same problem - the timelapse created is 48 bytes and (from what I can tell from the screenshots below), only a few frames long. However, the individual images for the timelapse - thousands of them - are actually created in a 'resized' subfolder as per the new timelapse.sh. 

I am using an ASI178MC if that's any clue, with no binning. Perhaps the images are too large? 

For tonight, I have set the binning to 2, and also reduced JPEG quality to 80. Will see if that works. My previous camera, ASI120MC, with the original timelapse.sh, works perfectly well in creating timelapses.

Thank you for any help!

[image]
[image]
[image]
[image]

thomasjacquin, 2020-10-19:
Try using the `scale` argument. https://github.com/thomasjacquin/allsky/issues/249#issuecomment-683332953
Just make sure to use the same ratio as the original images.

ebeyonder, 2020-10-19:
Thanks very much, as always, Thomas. 

I have now set binning to 2x2, so the images are now 1548 x 1040, which seem closer to the images produced by the ASI120 (1280 x 960). The file size per image also seems more manageable now at about 100KB or so, compared with 600KB before. I will see if this works tonight, if not I will try the scale argument. Thank you!

ebeyonder, 2020-10-20:
The binning didn't work, so I am now trying the scale argument for tonight. 

I went through the all sky.log and spotted this line:

Oct 20 06:53:52 allsky allsky.sh[2899]: Premature end of JPEG file

Oct 20 07:07:03 allsky allsky.sh[2899]: mv: Access failed: 550 Sorry, but that file doesn't exist (true)

Oct 20 07:08:08 allsky allsky.sh[2899]: ffmpeg version 4.1.6-1~deb10u1+rpt1 Copyright (c) 2000-2020 the FFmpeg developers
Oct 20 07:08:08 allsky allsky.sh[2899]:   built with gcc 8 (Raspbian 8.3.0-6+rpi1)

Oct 20 07:08:08 allsky allsky.sh[2899]:   WARNING: library configuration mismatch

Oct 20 07:10:54 allsky allsky.sh[2899]: cp: cannot create regular file 'images/20201020/image-20201020071054.jpg': File exists

There are other entries but these seem the most relevant.

Any help in figuring out what these mean, and the solution, would be most appreciated!

ebeyonder, 2020-10-20:
[image]

Tried the scale argument, but this is the error I got - no file created?

ebeyonder, 2020-10-20:
OK, I added a missing '\' and now it works. I am at binning 2x2, but used the scale command to scale it down to just 774x520 - well, it is smaller, but it works. I will try no binning and then use the scale to scale it down to half to see if that works. Thanks so much!

closed 2020-10-20 by ebeyonder