AllskyTeam/allsky #293: Camera setting in webportal blank

ctreptow65, 2020-11-28:
Hi All

I've installed allsky multiple times and everytime the liveview shows a broken icon and the camera setting only show the save and set default buttons. 

Setup
Raspberry Pi 3b+
ZWO 120MC

What I've tried so far

- Reinstalled via the instructions multiple times
- Re-Installed Raspian multiple times
- Checked /var/www/html and there is no settings.json file there after Gui intstall
- I can access the portal but the live view and camera settings do not work    

below is the readout from the allsky status command. 

allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-11-28 16:37:55 EST; 2min 35s ago
 Main PID: 759 (allsky.sh)
    Tasks: 7 (limit: 2063)
   CGroup: /system.slice/allsky.service
           ├─759 /bin/bash /home/pi/allsky/allsky.sh
           └─870 /home/pi/allsky/capture -angle -6 -autoexposure 1 -autogain 0 -bin 1 -brightness 50 -coolerEnabled 0 -darkframe 0 -daytimeDelay 5000 -delay 10 -exposure 10000 -filename image

Nov 28 16:40:24 allsky allsky.sh[759]: cp: missing destination file operand after 'liveview-.'
Nov 28 16:40:24 allsky allsky.sh[759]: Try 'cp --help' for more information.
Nov 28 16:40:29 allsky allsky.sh[759]: /home/pi/allsky/config.sh: line 69: /etc/raspap/autocam.sh: No such file or directory
Nov 28 16:40:29 allsky allsky.sh[759]: jq: error: Could not open file /etc/raspap/settings_auto.json: No such file or directory
Nov 28 16:40:29 allsky allsky.sh[759]: jq: error: Could not open file /etc/raspap/settings_auto.json: No such file or directory
Nov 28 16:40:29 allsky allsky.sh[759]: /home/pi/allsky/scripts/darkCapture.sh: line 16: [: =: unary operator expected
Nov 28 16:40:29 allsky allsky.sh[759]: jq: error: Could not open file /etc/raspap/settings_auto.json: No such file or directory
Nov 28 16:40:29 allsky allsky.sh[759]: scripts/saveImageDay.sh: line 12: [: =: unary operator expected
Nov 28 16:40:29 allsky allsky.sh[759]: cp: missing destination file operand after 'liveview-.'
Nov 28 16:40:29 allsky allsky.sh[759]: Try 'cp --help' for more information.
~
~

thomasjacquin, 2020-11-28:
@ctreptow65 The issue may have been caused by a recent change to the code. I will do a fresh install at home to see what is causing it. In the meantime, if you want to you can use this previous version of the code and follow the instructions as usual
https://github.com/thomasjacquin/allsky/archive/ae0e792eb2cde3fd881f9b7e42c6930f89253d5f.zip

ctreptow65, 2020-11-29:
Thanks that version works, awesome program

vidma, 2020-11-29:
that's weird/interesting. 
some of my guesses: 
- something might be wrong with directories setup, or permissions.  Probably `ALLSKY_HOME` env variable value is not consistent among the scripts `allsky.sh` and `config.sh` ?
- @ctreptow65 **did you used GUI to create the settings and/or config.sh files ? @thomasjacquin My guess is that then the `config.sh` would not contain the recent changes**, namely these lines https://github.com/thomasjacquin/allsky/blob/425c90716c4b9571c64c437f56023c0958ef9d63/config.sh.repo#L66-L70 , and we'd get kinda similar issue in the logs...
   * **so apparently someone needs to update the GUI tool as well!** @thomasjacquin 

also why do we have path different prefixes:
- /home/pi/allsky/config.sh
- /home/pi/allsky/allsky.sh - why  /home/pi/allsky and /etc/raspap/autocam.sh below ? do you have customly set directory somewhere?
- /etc/raspap/autocam.sh - not found, must be auto created by allsky.sh . is the directory writeable? see https://github.com/thomasjacquin/allsky/blob/425c90716c4b9571c64c437f56023c0958ef9d63/allsky.sh#L9 and  https://github.com/thomasjacquin/allsky/blob/425c90716c4b9571c64c437f56023c0958ef9d63/allsky.sh#L57 )
- /etc/raspap/settings_auto.json  -  this file should be never searched for (i.e. never used)  if autocam.sh existed, see https://github.com/thomasjacquin/allsky/blob/425c90716c4b9571c64c437f56023c0958ef9d63/config.sh.repo#L66-L70

ctreptow65, 2020-11-29:
Until I loaded the archived version I edited the config files in the terminal. I could not find the settings json so I edited the settings_ZWO.json file. I did edit the config.sh but only to set the upload the ftp to true. Once I loaded the archive I was editing within the GUI, 

Checking the logs after the 1st night of operation I can see I'm getting the ftp cert error, added the ssl-cert option to the process scripts this morning, but I am getting the allsky.sh [716] sunwait: not found

thomasjacquin, 2020-11-30:
@vidma I believe I have fixed the issue. See this commit https://github.com/thomasjacquin/allsky/commit/425c90716c4b9571c64c437f56023c0958ef9d63 and this commit on the GUI code https://github.com/thomasjacquin/allsky-portal/commit/d11ca1e43914c9e2786cace9413dce92e34ce528.
The issues were caused by the use of the wrong environment variable in `allsky.sh` and it created a side effect on this line https://github.com/thomasjacquin/allsky/blob/master/gui/install.sh#L61

It seems to work on my home camera. Let me know if you see a problem with those changes.

vidma, 2020-11-30:
yes, @thomasjacquin ,  makes sense.

ctreptow65, 2020-11-30:
Great job, re-installed via the instructions and now it works..

closed 2020-11-30 by ctreptow65