AllskyTeam/allsky #294: Unable to FTP to my webpage using AllSky Administration Panel

AstroHighlander, 2020-12-06:
I have browsed all the previous FTP-related issue threads and just can't work out a solution for myself.

I have successfully built & setup an AllSky camera which captures live view images & compiles them into time-lapse, star trail, & keogram at the end of the night.

I have also set up a website with webpage 'allsky' to which I am desperately trying to automatically FTP the relevant files but without success.

Using Filezilla I can refresh the pi directory to get the most recent image-resize.jpg file then double-click it over to my webpage and it appears no problem.

I just can't get it to work using the ftp-settings.sh script, and I've tried goodness knows how many permutations.

I've set the relevant config.sh parameters to 'true' so hopefully it's nothing as elementary as that.

I'm hoping for some "walk through" help here as I'm 63 years of age with absolutely no coding or website build experience - some of the comments & assistance on the other threads was way beyond my ability to understand so it really is "baby guidance" I'm asking for.

Hoping someone can help

Eric

AstroHighlander, 2020-12-15:
First of all, a big thanks to Thomas Jacquin who’s been helping me with this and achieved a great deal of progress so far and 

•	I can now FTP to my website page using the Pi terminal !!!

However, I’m still unable to automatically FTP anything to my website page via the AllSky frontend system

Here is the Pi terminal FTP script which works successfully:

lftp ftp://userxxxx:password@domain.co.uk -e "set net:max-retries 1; set net:timeout 20; set ssl:check-hostname no; cd /public_html/allsky/; put image-resize.jpg; bye"

Unfortunately, simply putting the appropriate terms into the ftp-settings.sh script doesn’t result in upload to my website.

Could anyone suggest which other scripts & files I need to review & amend so that this will work?

I’ve already trawled through other FTP threads and tried many of the offered solutions (including #74)

I would be very happy even if I could only achieve transfer of the live image (image-resize.jpg) as I’m sure I could work out what to do from there.

thomasjacquin, 2020-12-16:
You'll have to modify these lines 
https://github.com/thomasjacquin/allsky/blob/master/scripts/saveImageNight.sh#L61
https://github.com/thomasjacquin/allsky/blob/master/scripts/saveImageDay.sh#L52
https://github.com/thomasjacquin/allsky/blob/master/scripts/endOfNight.sh#L40
https://github.com/thomasjacquin/allsky/blob/master/scripts/endOfNight.sh#L53

courtois341, 2021-03-07:
Hello, having the same problems as the author of this post, I made the changes mentioned above and I therefore managed to upload the daily files to my site. Having failed to configure the ftp-setting.sh file I therefore do not have the files for the web page which are uploaded automatically with I only have the daily files which are uploaded (startrails, keograms, timelaps and image resize is ok! on the other hand the website does not exist! what should i do? 
I did a manual start ./allsky.sh and I have no errors appearing. yet I have no website.
thank you for your answers i am french and i do not speak english i go through a translator thank you for your understanding

dpoulson, 2021-03-19:
Edit /etc/lftp.conf and add a line that says:

set ssl:check-hostname no

No need to edit the scripts then.

courtois341, 2021-03-21:
I tried to edit the file
etc / lftpconf but writing is not allowed. what is the command to edit the file?
Thank you!

maserowik, 2021-03-21:
Try
sudo nano /etc/lftpconf

Michael A. Serowik

courtois341, 2021-03-22:
Thanks, I was able to modify the /etc/lftp.conf file by adding the text to it. Should I put it somewhere in the file or put it anywhere? I put it at the end of the file. I'm waiting to see if the website is created with my host. I will tell you if it works, Thank you for your help!

EricClaeys, 2021-04-26:
Adding the line to /etc/lftp.conf worked for me.  Sure beats editing the various .php files.